ch32-rs/wchisp #2: Please comment on "config_registers ... bit_range: [x, y] "  is 'x' included ?  (possible bug):

Pe3ucTop, 2022-05-16:
Hi, great work ! 
I looked for example of bit definition and get confused :
'''
      - bit_range: [7, 0]
        name: RDPR
        description: Read Protection. 0xA5 for unprotected, otherwise read-protected(ignoring WRP)
        explaination:
          0xa5: Unprotected
          _: Protected
'''
in compare to:
"
      - bit_range: [23, 21]
        name: SRAM_CODE_MODE
        description: SRAM Code Mode
        explaination:
          0b00: CODE-192KB + RAM-128KB
          0b01: CODE-224KB + RAM-96KB
          0b10: CODE-256KB + RAM-64KB
          0b11: CODE-228KB + RAM-32KB
"
So question: 'bit_range: [x , y]'   is 'x' included in range ? 
Is second part 3 bit or 2 bit ?
Which one is correct way?

andelf, 2022-05-16:
Thanks, it's a typo in offset calculation.

andelf, 2022-05-16:
It's an experiment in config register parsing. Currently, it uses `RangeInclusive` type to represent bit offsets.
I made a mistake in calculation.

commit 4c490c1c (2022-05-16): fix: typo in register offset

Close #2
closed 2022-05-16 by andelf