ch32-rs/wchisp #4: `Idenfity chip failed` on CH549 & CH552

glatavento, 2022-05-25:
Hi! Thank you for this project, which has offered me with a lot of convenience. 

I happen to have a CH552 board, a CH549 board, and CH32F103/V103/V307 boards, so I tested this tool on them. CH32F103/V103/V307 are working well, and here are the results of CH549 & CH552:

CH549:
```
12:21:29 [INFO] Found USB Device Bus 001 Device 046: ID 4348:55e0
12:21:29 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
12:21:29 [DEBUG] (1) wchisp::transport: <= a1c00200 4912
```

CH552:
```
12:21:49 [INFO] Found USB Device Bus 001 Device 048: ID 4348:55e0
12:21:49 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
12:21:49 [DEBUG] (1) wchisp::transport: <= a1ee0200 5211
```

As a comparison, here are the results of these two boards using isp55e0:
```
Found device CH549
Bootloader version 2.4.0
Unique chip ID fc-53-49-43
```
```
Found device CH552
Bootloader version 2.5.0
Unique chip ID 7f-7b-00-00
```

Here are the links where I bought them: [CH552](https://item.taobao.com/item.htm?id=669901203749) [CH549](https://item.taobao.com/item.htm?id=653726693686).

andelf, 2022-06-04:
Many thanks. I'll check this. 
I think I can use my WCH-Link(which is a CH549) for testing.

andelf, 2022-06-04:
The reason for fail chip detection is that CH549 and CH552's USB protocol differs from CH32Vx.

> a1ee0200 5211
> a1c00200 4912

The second byte is used for error detection on CH32Vx devices.
I'll try to figure out what it means for CH54x and CH55x families.

glatavento, 2022-06-06:
Thank you for your time. I made a ch554 board and had another try on it, and below is the output: 

```
04:41:15 [INFO] Found USB Device Bus 001 Device 020: ID 4348:55e0
04:41:15 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
04:41:15 [DEBUG] (1) wchisp::transport: <= a1bf0200 5411
Error: idenfity chip failed
```

isp55e0:
```
Found device CH554
Bootloader version 2.4.0
Unique chip ID ef-2e-5b-bc
```

Hope this helps.

BTW, `IOError` occurs really often during my tests on CH554/CH552/CH549 boards (about 9 in 10 times).

nikitos1550, 2022-06-06:
Hi! I have similar problem

```
> sudo ./wchisp info
15:15:18 [INFO] Found USB Device Bus 001 Device 064: ID 4348:55e0
15:15:18 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
15:15:18 [DEBUG] (1) wchisp::transport: <= a19b0200 4912
Error: idenfity chip failed
```

glatavento, 2022-06-06:
> > Thank you for your time. I made a ch554 board and had another try on it, and below is the output:
> > 
> > ```
> > 04:41:15 [INFO] Found USB Device Bus 001 Device 020: ID 4348:55e0
> > 04:41:15 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
> > 04:41:15 [DEBUG] (1) wchisp::transport: <= a1bf0200 5411
> > Error: idenfity chip failed
> > ```
> > 
> > isp55e0:
> > 
> > ```
> > Found device CH554
> > Bootloader version 2.4.0
> > Unique chip ID ef-2e-5b-bc
> > ```
> > 
> > Hope this helps.
> > 
> > BTW, `IOError` occurs really often during my tests on CH554/CH552/CH549 boards (about 9 in 10 times).
> 
> what is isp55e0 ?

https://github.com/frank-zago/isp55e0

nikitos1550, 2022-06-06:
> @glatavento @nikitos1550 Could you try with the newest code?

Yes, thanks. I will try later today or tomorrow and replay you.

nikitos1550, 2022-06-07:
Seems it is ok now:
```
./wchisp info
09:51:30 [INFO] Found USB Device Bus 001 Device 005: ID 4348:55e0
09:51:30 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
09:51:30 [DEBUG] (1) wchisp::transport: <= a19b0200 4912
09:51:30 [DEBUG] (1) wchisp::flashing: found chip: CH549[0x4912]
09:51:30 [DEBUG] (1) wchisp::transport: => a70200   1f00
09:51:30 [DEBUG] (1) wchisp::transport: <= a79b1a00 1f00000000002000000000d234c800020400f4b4b1400000a5f5
09:51:30 [DEBUG] (1) wchisp::flashing: read_config: 000000002000000000d234c800020400f4b4b1400000a5f5
09:51:30 [INFO] Chip: CH549[0x4912] (Code Flash: 60KiB, Data EEPROM: 1KiB)
09:51:30 [INFO] Chip UID: f4-b4-b1-40-00-00-a5-f5
09:51:30 [INFO] BTVER(bootloader ver): 02.40
09:51:30 [DEBUG] (1) wchisp::transport: => a70200   0700
09:51:30 [DEBUG] (1) wchisp::transport: <= a79b0e00 0700000000002000000000d234c8
REG0: 0x00000000
REG1: 0x00000020
  [7:0] REG1_0 0b100000 (0x20)
REG2: 0xC834D200
  [2:0] LV_RST_VOL 0b0 (0x0)
    `- 2.4V
  [8:8] MUST_0 0b0 (0x0)
  [9:9] MUST_1 0b1 (0x1)
  [12:12] En_P5.7_RESET 0b1 (0x1)
  [13:13] En_Long_Reset 0b0 (0x0)
  [14:14] No_Boot_Load 0b1 (0x1)
  [15:15] Code_Protect 0b1 (0x1)
```

```
./wchisp flash WCHLINK_V2.10.bin 
09:53:33 [INFO] Found USB Device Bus 001 Device 006: ID 4348:55e0
09:53:33 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
09:53:33 [DEBUG] (1) wchisp::transport: <= a19b0200 4912
09:53:33 [DEBUG] (1) wchisp::flashing: found chip: CH549[0x4912]
09:53:33 [DEBUG] (1) wchisp::transport: => a70200   1f00
09:53:33 [DEBUG] (1) wchisp::transport: <= a79b1a00 1f00000000002000000000d234c800020400f4b4b1400000a5f5
09:53:33 [DEBUG] (1) wchisp::flashing: read_config: 000000002000000000d234c800020400f4b4b1400000a5f5
09:53:33 [INFO] Chip: CH549[0x4912] (Code Flash: 60KiB, Data EEPROM: 1KiB)
09:53:33 [INFO] Chip UID: f4-b4-b1-40-00-00-a5-f5
09:53:33 [INFO] BTVER(bootloader ver): 02.40
09:53:33 [DEBUG] (1) wchisp::transport: => a70200   0700
09:53:33 [DEBUG] (1) wchisp::transport: <= a79b0e00 0700000000002000000000d234c8
REG0: 0x00000000
REG1: 0x00000020
  [7:0] REG1_0 0b100000 (0x20)
REG2: 0xC834D200
  [2:0] LV_RST_VOL 0b0 (0x0)
    `- 2.4V
  [8:8] MUST_0 0b0 (0x0)
  [9:9] MUST_1 0b1 (0x1)
  [12:12] En_P5.7_RESET 0b1 (0x1)
  [13:13] En_Long_Reset 0b0 (0x0)
  [14:14] No_Boot_Load 0b1 (0x1)
  [15:15] Code_Protect 0b1 (0x1)
09:53:33 [INFO] Read WCHLINK_V2.10.bin as Binary format
09:53:33 [INFO] Firmware size: 58368
09:53:33 [DEBUG] (1) wchisp::transport: => a40400   3a000000
09:53:38 [DEBUG] (1) wchisp::transport: <= a49b0200 0000
09:53:38 [INFO] Erased 58 code flash sectors
09:53:39 [DEBUG] (1) wchisp::transport: => a31e00   000000000000000000000000000000000000000000000000000000000000
09:53:39 [DEBUG] (1) wchisp::transport: <= a39b0200 e100
09:53:39 [DEBUG] (1) wchisp::transport: => a53d00   000000007d3135d2333333337c333333333333337c333333333333337c333333333333337c333333313f10337c333333313f18337c333333333333337c
09:53:39 [DEBUG] (1) wchisp::transport: <= a59b0200 0000
09:53:39 [DEBUG] (1) wchisp::transport: => a53d00   380000008e333333333333337c333333313f70337c333333333333337c333333333333337c333333333333337c333333333333337c333333333333337c
09:53:39 [DEBUG] (1) wchisp::transport: <= a59b0200 0000
[cut]
09:53:47 [DEBUG] (1) wchisp::transport: <= a69b0200 0000
09:53:47 [DEBUG] (1) wchisp::transport: => a61500   f0e300008f333333333333337c333333333333337c
09:53:47 [DEBUG] (1) wchisp::transport: <= a69b0200 0000
09:53:48 [INFO] Flash and Verify OK, now reset device
09:53:48 [DEBUG] (1) wchisp::transport: => a20100   01
```

andelf, 2022-06-09:
@nikitos1550  Thanks for your info.

closed 2022-11-13 by andelf