ch32-rs/wchisp #5: CH573F compatibility

nikitos1550, 2022-06-07:
Hello!

```
./wchisp info
10:26:21 [INFO] Found USB Device Bus 001 Device 018: ID 4348:55e0
10:26:21 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
10:26:21 [DEBUG] (1) wchisp::transport: <= a1000200 7313
10:26:21 [DEBUG] (1) wchisp::flashing: found chip: CH573[0x7313]
10:26:21 [DEBUG] (1) wchisp::transport: => a70200   1f00
10:26:21 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00a55affffffffffffff3fff6f000208000b5ddce4c284a9c6
10:26:21 [DEBUG] (1) wchisp::flashing: read_config: a55affffffffffffff3fff6f000208000b5ddce4c284a9c6
10:26:21 [INFO] Chip: CH573[0x7313] (Code Flash: 448KiB, Data EEPROM: 32KiB)
10:26:21 [INFO] Chip UID: 0b-5d-dc-e4-c2-84-a9-c6
10:26:21 [INFO] BTVER(bootloader ver): 02.80
10:26:21 [DEBUG] (1) wchisp::transport: => a70200   0700
10:26:21 [DEBUG] (1) wchisp::transport: <= a7000e00 0700a55affffffffffffff3fff6f
```

```
./wchisp unprotect
10:26:52 [INFO] Found USB Device Bus 001 Device 019: ID 4348:55e0
10:26:52 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
10:26:52 [DEBUG] (1) wchisp::transport: <= a1000200 7313
10:26:52 [DEBUG] (1) wchisp::flashing: found chip: CH573[0x7313]
10:26:52 [DEBUG] (1) wchisp::transport: => a70200   1f00
10:26:52 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00a55affffffffffffff3fff6f000208000b5ddce4c284a9c6
10:26:52 [DEBUG] (1) wchisp::flashing: read_config: a55affffffffffffff3fff6f000208000b5ddce4c284a9c6
10:26:52 [DEBUG] (1) wchisp::transport: => a70200   0700
10:26:52 [DEBUG] (1) wchisp::transport: <= a7000e00 0700a55affffffffffffff3fff6f
10:26:52 [DEBUG] (1) wchisp::transport: => a80d00   0700a55affffffffffffffffffff
10:26:52 [DEBUG] (1) wchisp::transport: <= a8000200 0000
10:26:52 [INFO] Code Flash unprotected
10:26:52 [DEBUG] (1) wchisp::transport: => a20100   01
10:26:52 [DEBUG] (1) wchisp::transport: <= a2000200 0000
10:26:52 [INFO] Device reset
```

```
 ./wchisp config
10:27:12 [INFO] Found USB Device Bus 001 Device 020: ID 4348:55e0
10:27:12 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
10:27:12 [DEBUG] (1) wchisp::transport: <= a1000200 7313
10:27:12 [DEBUG] (1) wchisp::flashing: found chip: CH573[0x7313]
10:27:12 [DEBUG] (1) wchisp::transport: => a70200   1f00
10:27:12 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00a55affffffffffffff3fff6f000208000b5ddce4c284a9c6
10:27:12 [DEBUG] (1) wchisp::flashing: read_config: a55affffffffffffff3fff6f000208000b5ddce4c284a9c6
10:27:12 [DEBUG] (1) wchisp::transport: => a70200   0700
10:27:12 [DEBUG] (1) wchisp::transport: <= a7000e00 0700a55affffffffffffff3fff6f
10:27:12 [INFO] Current config registers: a55affffffffffffff3fff6f
10:27:12 [INFO] Reset config registers:   a55affffffffffffff3fff6f
10:27:12 [DEBUG] (1) wchisp::transport: => a80d00   0700a55affffffffffffff3fff6f
10:27:12 [DEBUG] (1) wchisp::transport: <= a8000200 0000
10:27:12 [DEBUG] (1) wchisp::transport: => a70200   0700
10:27:12 [DEBUG] (1) wchisp::transport: <= a7000e00 0700a55affffffffffffff3fff6f
```

```
./wchisp erase
10:27:30 [INFO] Found USB Device Bus 001 Device 021: ID 4348:55e0
10:27:30 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
10:27:30 [DEBUG] (1) wchisp::transport: <= a1000200 7313
10:27:30 [DEBUG] (1) wchisp::flashing: found chip: CH573[0x7313]
10:27:30 [DEBUG] (1) wchisp::transport: => a70200   1f00
10:27:30 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00a55affffffffffffff3fff6f000208000b5ddce4c284a9c6
10:27:30 [DEBUG] (1) wchisp::flashing: read_config: a55affffffffffffff3fff6f000208000b5ddce4c284a9c6
10:27:30 [DEBUG] (1) wchisp::transport: => a40400   c0010000
10:27:30 [DEBUG] (1) wchisp::transport: <= a4000200 0000
10:27:30 [INFO] Erased 448 code flash sectors
```

```
/wchisp flash /home/nikita1550/seatbelt/wch/TMP/sample_central/firmware.hex 
10:28:51 [INFO] Found USB Device Bus 001 Device 025: ID 4348:55e0
10:28:51 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
10:28:51 [DEBUG] (1) wchisp::transport: <= a1000200 7313
10:28:51 [DEBUG] (1) wchisp::flashing: found chip: CH573[0x7313]
10:28:51 [DEBUG] (1) wchisp::transport: => a70200   1f00
10:28:51 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00a55affffffffffffff3fff6f000208000b5ddce4c284a9c6
10:28:51 [DEBUG] (1) wchisp::flashing: read_config: a55affffffffffffff3fff6f000208000b5ddce4c284a9c6
10:28:51 [INFO] Chip: CH573[0x7313] (Code Flash: 448KiB, Data EEPROM: 32KiB)
10:28:51 [INFO] Chip UID: 0b-5d-dc-e4-c2-84-a9-c6
10:28:51 [INFO] BTVER(bootloader ver): 02.80
10:28:51 [DEBUG] (1) wchisp::transport: => a70200   0700
10:28:51 [DEBUG] (1) wchisp::transport: <= a7000e00 0700a55affffffffffffff3fff6f
10:28:51 [INFO] Read /home/nikita1550/seatbelt/wch/TMP/sample_central/firmware.hex as IntelHex format
10:28:51 [INFO] Firmware size: 53248
10:28:51 [DEBUG] (1) wchisp::transport: => a40400   35000000
10:28:51 [DEBUG] (1) wchisp::transport: <= a4000200 0000
10:28:51 [INFO] Erased 53 code flash sectors
10:28:52 [DEBUG] (1) wchisp::transport: => a31e00   000000000000000000000000000000000000000000000000000000000000
10:28:52 [DEBUG] (1) wchisp::transport: <= a3000200 5b00
10:28:52 [DEBUG] (1) wchisp::transport: => a53d00   0000000008b2dd5dbbdddddd50ddddddddb2dd5d58b2dd9dc3dddddd50dddddddddddddd50dddddddddddddd50dddddddddddddd50ddddddddb2dd9d56
10:28:52 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
10:28:52 [DEBUG] (1) wchisp::transport: => a53d00   3800000014ddddddddb2dd3d55ddddddddb2dd5d55b2ddbdd8b2dd9d55b2ddfdd8b2dddd55b2dd3dd9b2dd1d54ddddddddb2ddbd54b2dd9dd9b2ddfd54
10:28:52 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
10:28:52 [DEBUG] (1) wchisp::transport: => a53d00   700000009cb2ddddd9b2dd1d4eb2dd1ddedddddd50b2ddbddeb2dd9d53b2ddfddeb2dddd53b2dd3ddfdc7ddcf0dc7ddc7ddc7ddcf0dc7ddc7ddc7ddcf0
10:28:52 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
10:28:52 [DEBUG] (1) wchisp::transport: => a53d00   a80000001cdc7ddc7ddc7ddcf0dc7ddc7ddc7ddcf0dc7ddc7ddc7ddcf06acadd9d4edbad55fe5d0ad9ceda5daafe5d3ad9de1b6a54ceab2bd07e581a54
10:28:52 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
10:28:52 [DEBUG] (1) wchisp::transport: => a53d00   e0000000dcfe5ddad9ceabd852f01bde1b7addccdad03ffe5d0ad9fed03ad9de1a7addce379addfe583add4e57dd96dcdddcdd204730226acadd9d4e56
10:28:52 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
[cut]
10:28:52 [DEBUG] (1) wchisp::transport: => a53500   d0cf000033dddddddddddddd50dddddddddddddd50dddddddddddddd50dddddddddddddd50dddddddddddddd50dddddddddddddd50
10:28:52 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
10:28:52 [DEBUG] (1) wchisp::transport: => a50500   00d00000af
10:28:52 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
10:28:52 [INFO] Code flash 53248 bytes written
10:28:54 [DEBUG] (1) wchisp::transport: => a31e00   000000000000000000000000000000000000000000000000000000000000
10:28:54 [DEBUG] (1) wchisp::transport: <= a3000200 5b00
10:28:54 [DEBUG] (1) wchisp::transport: => a63d00   000000007ab2dd5dbbdddddd50ddddddddb2dd5d58b2dd9dc3dddddd50dddddddddddddd50dddddddddddddd50dddddddddddddd50ddddddddb2dd9d56
10:28:54 [DEBUG] (1) wchisp::transport: <= a6000200 0e00
Error: Verify failed, mismatch
```

Questions:
1) Why verify fails?
2) How to burn dataflash?
3) How to use config registers (and how is it related to unprotect, if I want to protect my firmware) ?

nikitos1550, 2022-06-08:
Fw upload sometimes works ok
```
14:04:04 [INFO] Found USB Device Bus 020 Device 016: ID 4348:55e0
14:04:04 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
14:04:04 [DEBUG] (1) wchisp::transport: <= a1000200 7313
14:04:04 [DEBUG] (1) wchisp::flashing: found chip: CH573[0x7313]
14:04:04 [DEBUG] (1) wchisp::transport: => a70200   1f00
14:04:04 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00ffffffffffffffff4d0f0040000208009c38dae4c28438a2
14:04:04 [DEBUG] (1) wchisp::flashing: read_config: ffffffffffffffff4d0f0040000208009c38dae4c28438a2
14:04:04 [INFO] Chip: CH573[0x7313] (Code Flash: 448KiB, Data EEPROM: 32KiB)
14:04:04 [INFO] Chip UID: 9c-38-da-e4-c2-84-38-a2
14:04:04 [INFO] BTVER(bootloader ver): 02.80
14:04:04 [DEBUG] (1) wchisp::transport: => a70200   0700
14:04:04 [DEBUG] (1) wchisp::transport: <= a7000e00 0700ffffffffffffffff4d0f0040
14:04:04 [INFO] Read firmware.hex as IntelHex format
14:04:04 [INFO] Firmware size: 3072
14:04:04 [WARN] erase_code: set min number of erased sectors to 8
14:04:04 [DEBUG] (1) wchisp::transport: => a40400   08000000
14:04:04 [DEBUG] (1) wchisp::transport: <= a4000200 0000
14:04:04 [INFO] Erased 8 code flash sectors
14:04:05 [DEBUG] (1) wchisp::transport: => a31e00   000000000000000000000000000000000000000000000000000000000000
14:04:05 [DEBUG] (1) wchisp::transport: <= a3000200 0300
14:04:05 [DEBUG] (1) wchisp::transport: => a53d00   00000000fbddb2f2ddb2b2b225b2b2b2b2ddb2322dddb272acb2b2b225b2b2b2b2b2b2b225b2b2b2b2b2b2b225b2b2b2b2b2b2b225b2b2b2b2ddb2f223
14:04:05 [DEBUG] (1) wchisp::transport: <= a5000200 0000
14:04:05 [DEBUG] (1) wchisp::transport: => a53d00   3800000062b2b2b2b2ddb25220b2b2b2b2ddb23220ddb2d2b7ddb2f220ddb292b7ddb2b220ddb252b6ddb27221b2b2b2b2ddb2d221ddb2f2b6ddb29221
14:04:05 [DEBUG] (1) wchisp::transport: <= a5000200 0000
14:04:05 [DEBUG] (1) wchisp::transport: => a53d00   7000000064ddb2b292ddb25226ddb272b1b2b2b225ddb2d2b1ddb2f226ddb292b1ddb2b226ddb252b0b312b385b312b312b312b385b312b312b312b385
14:04:05 [DEBUG] (1) wchisp::transport: <= a5000200 0000
14:04:05 [DEBUG] (1) wchisp::transport: => a53d00   a80000005db312b312b312b385b312b312b312b385b312b312b312b38505a5b2f221b4c220913265b6a1b532df913255b6b1740521a1c444bf11377521
14:04:05 [DEBUG] (1) wchisp::transport: <= a5000200 0000
14:04:05 [DEBUG] (1) wchisp::transport: => a53d00   e0000000b39132b5b6a1c4b7278774b17415b2a3afaf50913265b691a555b6b17515b2a15245bda1d5f5b291a055b221b5b2f9b325b3b24fa55f4d0532
14:04:05 [DEBUG] (1) wchisp::transport: <= a5000200 0000
14:04:05 [DEBUG] (1) wchisp::transport: => a53d00   18010000bab2f221b4c2b791a565b6a1b5324891a555b6cf3b91261525b3b2b3b2b3b2b3259132b5b691326521913255b637f48502b2f211b1653291a5
14:04:05 [DEBUG] (1) wchisp::transport: <= a5000200 0000
14:04:05 [DEBUG] (1) wchisp::transport: => a53d00   50010000f8b5b63032a1c4b721df6cb17415b2f3afaf50913265b691a555b6b17515b2a15245bda1d5b5b391a055b221b5b2cfb325b3b24fa55f4d0532
14:04:05 [DEBUG] (1) wchisp::transport: <= a5000200 0000
14:04:05 [DEBUG] (1) wchisp::transport: => a53d00   8801000038b2f221b4c2b791a565b6a1b53248cfac913255b6a1d7b721912615b2b3b2b325b3b2b3b29132b521913265b6913255213ff48595b2f2c392
14:04:05 [DEBUG] (1) wchisp::transport: <= a5000200 0000
14:04:05 [DEBUG] (1) wchisp::transport: => a53d00   c0010000b4f3a333f4b3f43360b3f7b4749b9e0532b2f2a1b5c2b791a555b6a1b5324891a555b6b175d5b6a15245bda1d5a5b291a655b69132b5b60065
14:04:05 [DEBUG] (1) wchisp::transport: <= a5000200 0000
14:04:05 [DEBUG] (1) wchisp::transport: => a53d00   f801000067f3b33032f3a33363b3f433f7b3f7b4e34b9805a5b2f2a122c2b7913255b6a1223248913255b6cf72912455b0b175d521a1c545bda1d5a525
14:04:05 [DEBUG] (1) wchisp::transport: <= a5000200 0000
14:04:05 [DEBUG] (1) wchisp::transport: => a53d00   30020000c4913155b69132b521b3120ff581b74727b3b2cfa7c74f30a5f3a390769470b4e31836b3f6d1a5262500f290f620f6f3243032b7b6a1b7321b
14:04:05 [DEBUG] (1) wchisp::transport: <= a5000200 0000
14:04:05 [DEBUG] (1) wchisp::transport: => a53d00   68020000b7f0b6738df3326f928585b2f231f5f5652bf40f39d1336521b3a3b47c83f5d1ac55b4a3f5d124552711b5b3b225b7b2252137b7eb8585b265
14:04:05 [DEBUG] (1) wchisp::transport: <= a5000200 0000
14:04:05 [DEBUG] (1) wchisp::transport: => a53d00   a00200004d25b4b2b22134347d31f543b2b174b725214545bdd1aa442540f2b7d330323162e5f2303231f54325b1f435f22145452a0425913275b23162
14:04:05 [DEBUG] (1) wchisp::transport: <= a5000200 0000
14:04:05 [DEBUG] (1) wchisp::transport: => a53d00   d80200008843b2214545bd04b23175b5b291b6456531f543b237b521d245bd11b543b2ff9aa5b7b2b2a1b7b77625624d6d553272c5a1c747bd11b51325
14:04:05 [DEBUG] (1) wchisp::transport: <= a5000200 0000
14:04:05 [DEBUG] (1) wchisp::transport: => a53d00   100300002231f743b240f2a520b2b2a1b797e321d047bdb7d3a5614dfad5b2f16991b1b6a527f591b1463291271632303231b5d6a5517cb54c91b1b6a5
14:04:05 [DEBUG] (1) wchisp::transport: <= a5000200 0000
14:04:05 [DEBUG] (1) wchisp::transport: => a53d00   4803000006303231b5d63251ebb54cb1f7f63230a531b5d632517cb5db91b016323032f3349476f870fc72b4e3a1c547b91ff518ac9c3b27f6d1b44525
14:04:05 [DEBUG] (1) wchisp::transport: <= a5000200 0000
14:04:05 [DEBUG] (1) wchisp::transport: => a53d00   80030000faabf7df85d78d3f61fc37ff854feb4f31d12a86b300f21061a0fb30fbf3b330a5a1e7bbb3a1c7472af78d90bb7f05b334947eb47c05b6ba25
14:04:05 [DEBUG] (1) wchisp::transport: <= a5000200 0000
14:04:05 [DEBUG] (1) wchisp::transport: => a53d00   b80300008de385a7f7378dc312db859874178d806021c5a7b23b59a140a7b2a1c747bd406550f6b7d330324f315b4eb3f77f0d8b54946ef868fc6ae0f3
[... cut ...]
14:04:07 [DEBUG] (1) wchisp::transport: <= a6000200 0000
14:04:07 [DEBUG] (1) wchisp::transport: => a63500   d00b0000a8b2b2b2b2b2b2b225b2b2b2b2b2b2b225b2b2b2b2b2b2b225b2b2b2b2b2b2b225b2b2b2b2b2b2b225b2b2b2b2b2b2b225
14:04:07 [DEBUG] (1) wchisp::transport: <= a6000200 0000
14:04:08 [INFO] Flash and Verify OK, now reset device
14:04:08 [DEBUG] (1) wchisp::transport: => a20100   01
14:04:08 [DEBUG] (1) wchisp::transport: <= a2000200 0000
14:04:08 [INFO] Device reset
```

But I don't know reason, whe sometimes it doesn't work.

andelf, 2022-06-09:
It seems to work when the firmware size is small enough.
I guess it is because of the erase size. When erase size is large, it should sleep more time to await the erasing to be finished.

nikitos1550, 2022-06-09:
How sleep can be added? Maybe some patch required?

andelf, 2022-06-10:
I just added `--no-erase` to `flash` command, which might be helpful to test.
Could you try the `erase` then `flash --no-erase` combination?

nikitos1550, 2022-06-10:
> I just added `--no-erase` to `flash` command, which might be helpful to test. Could you try the `erase` then `flash --no-erase` combination?

Yes, I will try and report you.

nikitos1550, 2022-06-11:
```
```
> sudo time ./wchisp erase
15:53:51 [INFO] Found USB Device Bus 001 Device 076: ID 4348:55e0
15:53:51 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
15:53:51 [DEBUG] (1) wchisp::transport: <= a1000200 7313
15:53:51 [DEBUG] (1) wchisp::flashing: found chip: CH573[0x7313]
15:53:51 [DEBUG] (1) wchisp::transport: => a70200   1f00
15:53:51 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00a55affffffffffffff3fff6f000208000b5ddce4c284a9c6
15:53:51 [DEBUG] (1) wchisp::flashing: read_config: a55affffffffffffff3fff6f000208000b5ddce4c284a9c6
15:53:51 [DEBUG] (1) wchisp::transport: => a40400   c0010000
15:53:51 [DEBUG] (1) wchisp::transport: <= a4000200 0000
15:53:51 [INFO] Erased 448 code flash sectors
0.00user 0.00system 0:00.06elapsed 8%CPU (0avgtext+0avgdata 4108maxresident)k
0inputs+0outputs (0major+175minor)pagefaults 0swaps

HERE I POWERED OFF/POWER ON device, otherwise flash operation doesn't work at all.

> sudo time ./wchisp flash --no-erase firmware.hex 
15:54:04 [INFO] Found USB Device Bus 001 Device 077: ID 4348:55e0
15:54:04 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
15:54:04 [DEBUG] (1) wchisp::transport: <= a1000200 7313
15:54:04 [DEBUG] (1) wchisp::flashing: found chip: CH573[0x7313]
15:54:04 [DEBUG] (1) wchisp::transport: => a70200   1f00
15:54:04 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00a55affffffffffffff3fff6f000208000b5ddce4c284a9c6
15:54:04 [DEBUG] (1) wchisp::flashing: read_config: a55affffffffffffff3fff6f000208000b5ddce4c284a9c6
15:54:04 [INFO] Chip: CH573[0x7313] (Code Flash: 448KiB, Data EEPROM: 32KiB)
15:54:04 [INFO] Chip UID: 0b-5d-dc-e4-c2-84-a9-c6
15:54:04 [INFO] BTVER(bootloader ver): 02.80
15:54:04 [DEBUG] (1) wchisp::transport: => a70200   0700
15:54:04 [DEBUG] (1) wchisp::transport: <= a7000e00 0700a55affffffffffffff3fff6f
15:54:04 [INFO] Read firmware.hex as IntelHex format
15:54:04 [WARN] Skipping erase
15:54:04 [INFO] Firmware size: 53248
15:54:04 [DEBUG] (1) wchisp::transport: => a31e00   000000000000000000000000000000000000000000000000000000000000
15:54:04 [DEBUG] (1) wchisp::transport: <= a3000200 5b00
15:54:04 [DEBUG] (1) wchisp::transport: => a53d00   0000000089b2dd5dbbdddddd50ddddddddb2dd5d58b2dd9dc3dddddd50dddddddddddddd50dddddddddddddd50dddddddddddddd50ddddddddb2dd9d56
15:54:04 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
15:54:04 [DEBUG] (1) wchisp::transport: => a53d00   3800000053ddddddddb2dd3d55ddddddddb2dd5d55b2ddbdd8b2dd9d55b2ddfdd8b2dddd55b2dd3dd9b2dd1d54ddddddddb2ddbd54b2dd9dd9b2ddfd54
15:54:04 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
15:54:04 [DEBUG] (1) wchisp::transport: => a53d00   70000000b1b2ddddd9b2dd1d4eb2dd1ddedddddd50b2ddbddeb2dd9d53b2ddfddeb2dddd53b2dd3ddfdc7ddcf0dc7ddc7ddc7ddcf0dc7ddc7ddc7ddcf0
15:54:04 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
15:54:04 [DEBUG] (1) wchisp::transport: => a53d00   a800000010dc7ddc7ddc7ddcf0dc7ddc7ddc7ddcf0dc7ddc7ddc7ddcf06acadd9d4edbad55fe5d0ad9ceda5daafe5d3ad9de1b6a54ceab2bd07e581a54
15:54:04 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
15:54:04 [DEBUG] (1) wchisp::transport: => a53d00   e000000043fe5ddad9ceabd852f01bde1b7addccdad03ffe5d0ad9fed03ad9de1a7addce379addfe583add4e57dd96dcdddcdd204730226acadd9d4e56
15:54:04 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
15:54:04 [DEBUG] (1) wchisp::transport: => a53d00   180100001aadd8fe5d0ad9ce575d27fe5d3ad9a0d9fe497adddcdddc50dcdddcddfe5dda54fe5d0ad9fe5d3a544edbcdd8eafadd107ede0a5dfe5dda54
15:54:04 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
15:54:04 [DEBUG] (1) wchisp::transport: => a53d00   50010000685f5dceabd8d9b08ede1b7add9c57d0b2fe5d0ad9fe5d3a54de1a7addcebada51fe583add4edadd2ddcdddcdd20ca30af6acadd9d4edbad55
15:54:04 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
15:54:04 [DEBUG] (1) wchisp::transport: => a53d00   8801000006fe5d0ad9ceda5daaa054fe5d3ad9ce35d8d9fe497adddc50dcdddcdddcddfed0dad9fe5d0ad9fed03ad94edbedd8ea77dd9da46a9ccc5c16
15:54:04 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
15:54:04 [DEBUG] (1) wchisp::transport: => a53d00   c00100008fdc9b5c98cc98db9670ff6acadd9dce57add8fe5d3ad9ce575d27fe5d3ad9de97bad9cebacaddfed33ad9fe5ddad96f109cdc5f5d5c9bdc16
15:54:04 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
15:54:04 [DEBUG] (1) wchisp::transport: => a53d00   f8010000595c98cc98c0f76a47dd9dcedaadd8fed03ad9ceda5d27fed03ad9a08afe4b3a52de1abad9cebaca50fe5e3ad9fe5dda54dc7d609aeed82852
15:54:04 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
15:54:04 [DEBUG] (1) wchisp::transport: => a53d00   30020000dfdcdda0c8a8205fd09cccff19fb1fdb967759dc99beca49506f9dff994f999c515f5dd8d9ced85d6e9fd91ce29c5d00e76acadd9dcedadd53
15:54:04 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
15:54:04 [DEBUG] (1) wchisp::transport: => a53d00   6802000037fe553adf589afef82c5daeddfdedfe53d95dc89afede39d0dcddfedf795d5fd05edab95d3e13daaefeded95d5f5d5e57b95d3e13da23de15
15:54:04 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
15:54:04 [DEBUG] (1) wchisp::transport: => a53d00   a002000004995d5f5d5edab9d03e13da23fedf79d05f5d9cccfb199792931ddb1bceaa285b709a7754f3544814bedb2addc498b867b8e2509993589867
15:54:04 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
15:54:04 [DEBUG] (1) wchisp::transport: => a53d00   d80200002e208420c9be45e9516f9d7f99cf945f199cdc5f5dce88d451cea828d298e2ff59106adcccfb11db9e6ad9d5dd8ceac81560eaaceab4eaf796
15:54:04 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
[... cut ...]
15:54:07 [DEBUG] (1) wchisp::transport: => a63d00   00000000e8b2dd5dbbdddddd50ddddddddb2dd5d58b2dd9dc3dddddd50dddddddddddddd50dddddddddddddd50dddddddddddddd50ddddddddb2dd9d56
15:54:07 [DEBUG] (1) wchisp::transport: <= a6000200 0e00
Error: Verify failed, mismatch
Command exited with non-zero status 1
0.02user 0.01system 0:02.14elapsed 1%CPU (0avgtext+0avgdata 4376maxresident)k
0inputs+0outputs (0major+283minor)pagefaults 0swaps
```

Now verify fails, this can be because of (as I remember) fact, that before I used winwchisp app for windows and locked firmware read bit. Will try get windows machine and check this (and unlock if locked) and try again.

nikitos1550, 2022-06-11:
Here are screenshots of original vendor win isp tool:
[image]
[image]
*Seems images are not rendered here, so you can follow links*

Without checked protect code item, programming fails. There is no interesting log, but maybe can be helpful for you.

nikitos1550, 2022-06-11:
I try another firmware that is much smaller then previous
```
> sudo ./wchisp erase  
16:41:21 [INFO] Found USB Device Bus 001 Device 082: ID 4348:55e0
16:41:21 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
16:41:21 [DEBUG] (1) wchisp::transport: <= a1000200 7313
16:41:21 [DEBUG] (1) wchisp::flashing: found chip: CH573[0x7313]
16:41:21 [DEBUG] (1) wchisp::transport: => a70200   1f00
16:41:21 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00fffffffffeffffffc73fff6f000208000b5ddce4c284a9c6
16:41:21 [DEBUG] (1) wchisp::flashing: read_config: fffffffffeffffffc73fff6f000208000b5ddce4c284a9c6
16:41:21 [DEBUG] (1) wchisp::transport: => a40400   c0010000
16:41:21 [DEBUG] (1) wchisp::transport: <= a4000200 0000
16:41:21 [INFO] Erased 448 code flash sectors

> sudo ./wchisp flash --no-erase firmware.hex 
16:41:26 [INFO] Found USB Device Bus 001 Device 083: ID 4348:55e0
16:41:26 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
16:41:26 [DEBUG] (1) wchisp::transport: <= a1000200 7313
16:41:26 [DEBUG] (1) wchisp::flashing: found chip: CH573[0x7313]
16:41:26 [DEBUG] (1) wchisp::transport: => a70200   1f00
16:41:26 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00fffffffffeffffffc73fff6f000208000b5ddce4c284a9c6
16:41:26 [DEBUG] (1) wchisp::flashing: read_config: fffffffffeffffffc73fff6f000208000b5ddce4c284a9c6
16:41:26 [INFO] Chip: CH573[0x7313] (Code Flash: 448KiB, Data EEPROM: 32KiB)
16:41:26 [INFO] Chip UID: 0b-5d-dc-e4-c2-84-a9-c6
16:41:26 [INFO] BTVER(bootloader ver): 02.80
16:41:26 [DEBUG] (1) wchisp::transport: => a70200   0700
16:41:26 [DEBUG] (1) wchisp::transport: <= a7000e00 0700fffffffffeffffffc73fff6f
16:41:26 [INFO] Read firmware.hex as IntelHex format
16:41:26 [WARN] Skipping erase
16:41:26 [INFO] Firmware size: 3072
16:41:26 [DEBUG] (1) wchisp::transport: => a31e00   000000000000000000000000000000000000000000000000000000000000
16:41:26 [DEBUG] (1) wchisp::transport: <= a3000200 5b00
16:41:26 [DEBUG] (1) wchisp::transport: => a53d00   0000000065b2dd9db2dddddd50ddddddddb2dd5d58b2dd9dc3dddddd50dddddddddddddd50dddddddddddddd50dddddddddddddd50ddddddddb2dd9d56
16:41:26 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
16:41:26 [DEBUG] (1) wchisp::transport: => a53d00   3800000091ddddddddb2dd3d55ddddddddb2dd5d55b2ddbdd8b2dd9d55b2ddfdd8b2dddd55b2dd3dd9b2dd1d54ddddddddb2ddbd54b2dd9dd9b2ddfd54
16:41:26 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
[cut]
16:41:26 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
16:41:26 [DEBUG] (1) wchisp::transport: => a50500   000c00009f
16:41:26 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
16:41:26 [INFO] Code flash 3072 bytes written
16:41:28 [DEBUG] (1) wchisp::transport: => a31e00   000000000000000000000000000000000000000000000000000000000000
16:41:28 [DEBUG] (1) wchisp::transport: <= a3000200 5b00
16:41:28 [DEBUG] (1) wchisp::transport: => a63d00   000000005fb2dd9db2dddddd50ddddddddb2dd5d58b2dd9dc3dddddd50dddddddddddddd50dddddddddddddd50dddddddddddddd50ddddddddb2dd9d56
16:41:28 [DEBUG] (1) wchisp::transport: <= a6000200 0e00
Error: Verify failed, mismatch

> sudo ./wchisp verify firmware.hex 
16:41:33 [INFO] Found USB Device Bus 001 Device 084: ID 4348:55e0
16:41:33 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
16:41:33 [DEBUG] (1) wchisp::transport: <= a1000200 7313
16:41:33 [DEBUG] (1) wchisp::flashing: found chip: CH573[0x7313]
16:41:33 [DEBUG] (1) wchisp::transport: => a70200   1f00
16:41:33 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00fffffffffeffffffc73fff6f000208000b5ddce4c284a9c6
16:41:33 [DEBUG] (1) wchisp::flashing: read_config: fffffffffeffffffc73fff6f000208000b5ddce4c284a9c6
16:41:33 [INFO] Read firmware.hex as IntelHex format
16:41:33 [INFO] Firmware size: 2588
16:41:34 [DEBUG] (1) wchisp::transport: => a31e00   000000000000000000000000000000000000000000000000000000000000
16:41:34 [DEBUG] (1) wchisp::transport: <= a3000200 5b00
16:41:34 [DEBUG] (1) wchisp::transport: => a63d00   00000000ceb2dd9db2dddddd50ddddddddb2dd5d58b2dd9dc3dddddd50dddddddddddddd50dddddddddddddd50dddddddddddddd50ddddddddb2dd9d56
16:41:34 [DEBUG] (1) wchisp::transport: <= a6000200 0e00
Error: Verify failed, mismatch
```

I can see that firmware size is different according tool log for flash and verify. Why so?

nikitos1550, 2022-06-13:
also 
```
10:34:15 [DEBUG] (1) wchisp::transport: => a53d00   0000000077b2cd9dfedddddd50ddddddddb2dd5d58b2dd9dc3dddddd50dddddddddddddd50dddddddddddddd50dddddddddddddd50ddddddddb2dd9d56
10:34:15 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
10:34:15 [DEBUG] (1) wchisp::transport: => a53d00   3800000077ddddddddb2dd3d55ddddddddb2dd7d7db2dd1df2b2dd9d55b2ddfdd8b2dddd55b2dd3dd9b2dd1d54ddddddddb2ddbd54b2dd9dd9b2ddfd54
10:34:15 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
10:34:15 [DEBUG] (1) wchisp::transport: => a53d00   700000007cb2ddddd9b2dd1d4eb2dd1ddedddddd50b2ddbddeb2dd9d53b2ddfddeb2dddd53b2dd3ddfdc7ddcf0dc7ddc7ddc7ddcf0dc7ddc7ddc7ddcf0
10:34:15 [DEBUG] (1) wchisp::transport: <= a5000200 fe00
```
is response code fe for flash write request OK?

nikitos1550, 2022-06-13:
@andelf in short for 0xA5 0xA6 commands last byte should be calculated proper way (I don't know how).
Here https://github.com/frank-zago/isp55e0/issues/2#issuecomment-1153918256 you can check dump made from official WCH tool. 
Also for your app, for sure you should check response code (I mentioned this in prev message).

andelf, 2022-11-13:
@nikitos1550  `fe 00` indicates errors. It might be writing outside the flash range.

andelf, 2022-11-16:
Confirm that CH573 works.

```console
> cargo run -- flash ./broadcaster.hex
    Finished dev [unoptimized + debuginfo] target(s) in 0.03s
     Running `target/debug/wchisp flash ./broadcaster.hex`
11:25:18 [INFO] Chip: CH573[0x7313] (Code Flash: 448KiB, Data EEPROM: 32KiB)
11:25:18 [INFO] Chip UID: fd-27-a2-26-3b-38-da-86
11:25:18 [INFO] BTVER(bootloader ver): 02.80
11:25:18 [INFO] Current config registers: ffffffffffffffff4d0fff4f
RESERVED: 0xFFFFFFFF
WPROTECT: 0xFFFFFFFF
  [0:0]   NO_KEY_SERIAL_DOWNLOAD 0x1 (0b1)
    `- Enable
  [1:1]   DOWNLOAD_CFG 0x1 (0b1)
    `- PB22(Default set)
USER_CFG: 0x4FFF0F4D
  [2:0]   RESERVED 0x5 (0b101)
    `- Default
  [3:3]   CFG_RESET_EN 0x1 (0b1)
    `- Enable
  [4:4]   CFG_DEBUG_EN 0x0 (0b0)
    `- Disable
  [5:5]   RESERVED 0x0 (0b0)
    `- Default
  [6:6]   CFG_BOOT_EN 0x1 (0b1)
    `- Enable
  [7:7]   CFG_ROM_READ 0x0 (0b0)
    `- Disable the programmer to read out, and keep the program secret
  [27:8]  RESERVED 0xFFF0F (0b11111111111100001111)
    `- Default
  [31:28] VALID_SIG 0x4 (0b100)
    `- Valid
11:25:18 [INFO] Read ./broadcaster.hex as IntelHex format
11:25:18 [INFO] Firmware size: 93184
11:25:18 [INFO] Erasing...
11:25:18 [INFO] Erased 92 code flash sectors
11:25:19 [INFO] Erase done
11:25:19 [INFO] Writing to code flash...
11:25:20 [INFO] Code flash 93184 bytes written
11:25:21 [INFO] Verifying...
11:25:22 [INFO] Verify OK
11:25:22 [INFO] Now reset device and skip any communication errors
11:25:22 [INFO] Device reset
```

closed 2024-11-02 by nikitos1550