ch32-rs/wchisp #6: isp checksum check not valid for CH569

hansfbaier, 2022-07-17:
For CH569, it reports isp_checksum failed,
but if I comment out those checks, the flashing works fine,
and the binary runs well.
Here is the log (with the first checksum check commented out):
```
$ ./target/debug/wchisp flash CH372Device.bin
19:38:53 [INFO] Found USB Device Bus 007 Device 016: ID 4348:55e0
19:38:53 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
19:38:53 [DEBUG] (1) wchisp::transport: <= a1000200 6910
19:38:53 [DEBUG] (1) wchisp::flashing: found chip: CH569[0x6910]
19:38:53 [DEBUG] (1) wchisp::transport: => a70200   1f00
19:38:53 [DEBUG] (1) wchisp::transport: <= a7001a00 1f0011bff9f713bff9ec45f2ff8f0002070032e7bae4c284ae50
19:38:53 [DEBUG] (1) wchisp::flashing: read_config: 11bff9f713bff9ec45f2ff8f0002070032e7bae4c284ae50
19:38:53 [INFO] Chip: CH569[0x6910] (Code Flash: 448KiB, Data EEPROM: 32KiB)
19:38:53 [INFO] Chip UID: 32-e7-ba-e4-c2-84-ae-50
19:38:53 [INFO] BTVER(bootloader ver): 02.70
19:38:53 [DEBUG] (1) wchisp::transport: => a70200   0700
19:38:53 [DEBUG] (1) wchisp::transport: <= a7000e00 070011bff9f713bff9ec45f2ff8f
19:38:53 [INFO] Read CH372Device.bin as Binary format
19:38:53 [DEBUG] (1) wchisp::transport: => a40400   2b000000
19:38:53 [DEBUG] (1) wchisp::transport: <= a4000200 0000
19:38:53 [INFO] Erased 43 code flash sectors
19:38:54 [INFO] Erase done
19:38:54 [INFO] Firmware size: 43008
19:38:54 [DEBUG] (1) wchisp::transport: => a31e00   000000000000000000000000000000000000000000000000000000000000
19:38:54 [DEBUG] (1) wchisp::transport: <= a3000200 0000
key_checksum: 41 payload checksum: 0
19:38:54 [DEBUG] (1) wchisp::transport: => a53d00   000000004994ebab91fbfbfb64fbfbfbfb94eb8b0294ebab9dea440293fbfbfbfbfbfbfb64fbfbfbfbfbfbfb64fbfbfbfbfbfbfb64fbfbfbfb94ebcb00
19:38:54 [DEBUG] (1) wchisp::transport: <= a5000200 0000
19:38:54 [DEBUG] (1) wchisp::transport: => a53d00   3800000033fbfbfbfb94eb2b07fbfbfbfb94eb8b0794fb0bd194ebcb0794ebeb9894fb4b4d94fb2bc994eb4b0694eb6b9994eb8b0694eb4bc394ebcb06
19:38:54 [DEBUG] (1) wchisp::transport: <= a5000200 0000
19:38:54 [DEBUG] (1) wchisp::transport: => a53d00   70000000f294ebeb9994eb0b0594eb2b9a94eb4b0594eb6b9a94eb8b0594ebab9a94ebcb0594ebeb9a94eb0b04fbfbfbfbfbfbfb64fbfbfbfbfbfbfb64
[...]
c8a700003efbfbfbfbfbfbfb64fbfbfbfbfbfbfb64fbfbfbfbfbfbfb64fbfbfbfbfbfbfb64fbfbfbfbfbfbfb64fbfbfbfbfbfbfb64fbfbfbfbfbfbfb64
19:38:54 [DEBUG] (1) wchisp::transport: <= a5000200 0000
19:38:54 [DEBUG] (1) wchisp::transport: => a50500   00a800006d
19:38:54 [DEBUG] (1) wchisp::transport: <= a5000200 0000
19:38:54 [INFO] Code flash 43008 bytes written
19:38:56 [DEBUG] (1) wchisp::transport: => a31e00   000000000000000000000000000000000000000000000000000000000000
19:38:56 [DEBUG] (1) wchisp::transport: <= a3000200 0000
Error: isp_key checksum failed

```

hansfbaier, 2022-07-17:
If I comment out the following lines, the tool works just fine for CH569:
```diff
diff --git a/src/flashing.rs b/src/flashing.rs
index ab6f6dc..0c9b4cd 100644
--- a/src/flashing.rs
+++ b/src/flashing.rs
@@ -159,7 +159,8 @@ impl<T: Transport> Flashing<T> {
         let isp_key = Command::isp_key(vec![0; 0x1e]);
         let resp = self.transport.transfer(isp_key)?;
         anyhow::ensure!(resp.is_ok(), "isp_key failed");
-        anyhow::ensure!(resp.payload()[0] == key_checksum, "isp_key checksum failed");
+        println!("key_checksum: {:x?} payload checksum: {:x?}", key_checksum, resp.payload()[0]);
+        // anyhow::ensure!(resp.payload()[0] == key_checksum, "isp_key checksum failed");
 
         const CHUNK: usize = 56;
         let mut address = 0x0;
@@ -184,7 +185,7 @@ impl<T: Transport> Flashing<T> {
         let isp_key = Command::isp_key(vec![0; 0x1e]);
         let resp = self.transport.transfer(isp_key)?;
         anyhow::ensure!(resp.is_ok(), "isp_key failed");
-        anyhow::ensure!(resp.payload()[0] == key_checksum, "isp_key checksum failed");
+        //anyhow::ensure!(resp.payload()[0] == key_checksum, "isp_key checksum failed");
 
         const CHUNK: usize = 56;
         let mut address = 0x0;
```

hansfbaier, 2022-07-17:
The program is flashed and verified successfully and the target resets.

bvernoux, 2022-07-17:
Becareful also the flash code sector size is 256 Bytes on WCH CH56x and not 1024 Bytes like for other CH32VXXX

elfmimi, 2022-08-10:
This is where **bvernoux** added support for the newer version of the bootloader which come with CH569 .

https://github.com/jmaselbas/wch-isp/commit/9fdfcbb35f2724a25bb8871f421f1c5be9543d87

capiman, 2024-01-09:
I just hit the same problem with a brand new CH569. 

$ wchisp -v flash build/HydraUSB3_USB.bin
07:53:00 [DEBUG] (1) wchisp::transport::usb: Found USB Device Bus 003 Device 005: ID 4348:55e0
07:53:00 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
07:53:00 [DEBUG] (1) wchisp::transport: <= a1000200 6910
07:53:00 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
07:53:00 [DEBUG] (1) wchisp::transport: <= a1000200 6910
07:53:00 [DEBUG] (1) wchisp::flashing: found chip: CH569[0x6910]
07:53:00 [DEBUG] (1) wchisp::transport: => a70200   1f00
07:53:00 [DEBUG] (1) wchisp::transport: <= a7001a00 1f0011bff9f713bff9ec45f2ffcf000207001c983b263b3892f6
07:53:00 [DEBUG] (1) wchisp::flashing: read_config: 11bff9f713bff9ec45f2ffcf000207001c983b263b3892f6
07:53:00 [INFO] Chip: CH569[0x6910] (Code Flash: 448KiB, Data EEPROM: 32KiB)
07:53:00 [INFO] Chip UID: 1C-98-3B-26-3B-38-92-F6
07:53:00 [INFO] BTVER(bootloader ver): 02.70
07:53:00 [DEBUG] (1) wchisp::transport: => a70200   1f00
07:53:00 [DEBUG] (1) wchisp::transport: <= a7001a00 1f0011bff9f713bff9ec45f2ffcf000207001c983b263b3892f6
07:53:00 [INFO] Current config registers: 11bff9f713bff9ec45f2ffcf000207001c983b263b3892f6
UNKNOWN0: 0xF7F9BF11
UNKNOWN1: 0xECF9BF13
NV_INFO: 0xCFFFF245
  [31:30] USER_MEM 0x3 (0b11)
    `- RAMX 96KB + ROM 32KB
  [29:29] LOCKUP_RST_EN 0x0 (0b0)
    `- NotReset
  [28:28] RESERVED1 0x0 (0b0)
    `- Reserved
  [27:12] RESERVED2 0xFFFF (0b1111111111111111)
    `- Reserved
  [11:10] RESERVED3 0x0 (0b0)
    `- Reserved
  [9:8]   RESERVED4 0x2 (0b10)
    `- Reserved
  [7:7]   CODE_READ_EN 0x0 (0b0)
    `- Disable
  [6:6]   BOOT_EN 0x1 (0b1)
    `- Enable
  [5:5]   DEBUG_EN 0x0 (0b0)
    `- Disable
  [4:4]   RESET_EN 0x0 (0b0)
    `- Disable, PB15 is used as GPIO
  [3:0]   RESERVED5 0x5 (0b101)
    `- Reserved
07:53:00 [INFO] Read build/HydraUSB3_USB.bin as Binary format
07:53:00 [INFO] Firmware size: 22528
07:53:00 [INFO] Erasing...
07:53:00 [DEBUG] (1) wchisp::transport: => a40400   17000000
07:53:00 [DEBUG] (1) wchisp::transport: <= a4000200 0000
07:53:00 [INFO] Erased 23 code flash sectors
07:53:01 [INFO] Erase done
07:53:01 [INFO] Writing to code flash...
07:53:01 [DEBUG] (1) wchisp::transport: => a31e00   000000000000000000000000000000000000000000000000000000000000
07:53:01 [DEBUG] (1) wchisp::transport: <= a3000200 0000
Error: isp_key checksum failed
$