ch32-rs/wchisp #9: cannot install wchisp under macOS?

wankey, 2022-08-09:
cargo install wchisp --git https://github.com/ch32-rs/wchisp
    Updating git repository `https://github.com/ch32-rs/wchisp`
  Installing wchisp v0.1.3 (https://github.com/ch32-rs/wchisp#aedf69aa)
    Updating crates.io index
  Downloaded time v0.3.13
  Downloaded 1 crate (92.6 KB) in 0.62s
   Compiling libc v0.2.127
   Compiling proc-macro2 v1.0.43
   Compiling quote v1.0.21
   Compiling unicode-ident v1.0.3
   Compiling syn v1.0.99
   Compiling version_check v0.9.4
   Compiling cfg-if v1.0.0
   Compiling autocfg v1.1.0
   Compiling pkg-config v0.3.25
   Compiling cc v1.0.73
   Compiling serde_derive v1.0.143
   Compiling log v0.4.17
   Compiling hashbrown v0.12.3
   Compiling memchr v2.5.0
   Compiling serde v1.0.143
   Compiling itoa v1.0.3
   Compiling heck v0.4.0
   Compiling linked-hash-map v0.5.6
   Compiling os_str_bytes v6.2.0
   Compiling time-macros v0.2.4
   Compiling anyhow v1.0.60
   Compiling rusb v0.9.1
   Compiling termcolor v1.1.3
   Compiling ppv-lite86 v0.2.16
   Compiling ryu v1.0.11
   Compiling textwrap v0.15.0
   Compiling once_cell v1.13.0
   Compiling bitflags v1.3.2
   Compiling strsim v0.10.0
   Compiling hex v0.4.3
   Compiling bitfield v0.13.2
   Compiling ihex v3.0.0
   Compiling scroll v0.11.0
   Compiling yaml-rust v0.4.5
   Compiling clap_lex v0.2.4
   Compiling proc-macro-error-attr v1.0.4
   Compiling proc-macro-error v1.0.4
   Compiling indexmap v1.9.1
   Compiling object v0.28.4
   Compiling getrandom v0.2.7
   Compiling num_threads v0.1.6
   Compiling atty v0.2.14
   Compiling libusb1-sys v0.6.4
   Compiling rand_core v0.6.3
   Compiling rand_chacha v0.3.1
   Compiling time v0.3.13
The following warnings were emitted during compilation:

warning: libusb/libusb/os/darwin_usb.c:23:10: fatal error: 'assert.h' file not found
warning: #include <assert.h>
warning:          ^~~~~~~~~~
warning: 1 error generated.

error: failed to run custom build command for `libusb1-sys v0.6.4`

Caused by:
  process didn't exit successfully: `/var/folders/4y/h7ws2sw13n36hvn3x20c89hm0000gn/T/cargo-installL8TD6a/release/build/libusb1-sys-f46848d315a8dfad/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-env-changed=LIBUSB_STATIC
  cargo:vendored=1
  cargo:static=1
  cargo:include=/var/folders/4y/h7ws2sw13n36hvn3x20c89hm0000gn/T/cargo-installL8TD6a/release/build/libusb1-sys-892fc22518a7ef9c/out/include
  cargo:rustc-link-lib=framework=CoreFoundation
  cargo:rustc-link-lib=framework=IOKit
  cargo:rustc-link-lib=framework=Security
  cargo:rerun-if-env-changed=LIBUDEV_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64-apple-darwin
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64_apple_darwin
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=LIBUDEV_STATIC
  cargo:rerun-if-env-changed=LIBUDEV_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-apple-darwin
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_apple_darwin
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-apple-darwin
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_apple_darwin
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-apple-darwin
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_apple_darwin
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR
  Including posix!
  TARGET = Some("x86_64-apple-darwin")
  OPT_LEVEL = Some("3")
  HOST = Some("x86_64-apple-darwin")
  CC_x86_64-apple-darwin = None
  CC_x86_64_apple_darwin = None
  HOST_CC = None
  CC = None
  CFLAGS_x86_64-apple-darwin = None
  CFLAGS_x86_64_apple_darwin = None
  HOST_CFLAGS = None
  CFLAGS = None
  CRATE_CC_NO_DEFAULTS = None
  DEBUG = Some("false")
  CARGO_CFG_TARGET_FEATURE = Some("fxsr,sse,sse2,sse3,ssse3")
  running: "cc" "-O3" "-ffunction-sections" "-fdata-sections" "-fPIC" "-m64" "-arch" "x86_64" "-I" "/var/folders/4y/h7ws2sw13n36hvn3x20c89hm0000gn/T/cargo-installL8TD6a/release/build/libusb1-sys-892fc22518a7ef9c/out/include" "-I" "libusb/libusb" "-Wall" "-Wextra" "-DPRINTF_FORMAT(a, b)=" "-DENABLE_LOGGING=1" "-DOS_DARWIN=1" "-DTARGET_OS_OSX=1" "-DHAVE_SYS_TIME_H=1" "-DHAVE_NFDS_T=1" "-DPLATFORM_POSIX=1" "-DHAVE_CLOCK_GETTIME=1" "-DDEFAULT_VISIBILITY=__attribute__((visibility(\"default\")))" "-o" "/var/folders/4y/h7ws2sw13n36hvn3x20c89hm0000gn/T/cargo-installL8TD6a/release/build/libusb1-sys-892fc22518a7ef9c/out/libusb/libusb/os/darwin_usb.o" "-c" "libusb/libusb/os/darwin_usb.c"
  cargo:warning=libusb/libusb/os/darwin_usb.c:23:10: fatal error: 'assert.h' file not found
  cargo:warning=#include <assert.h>
  cargo:warning=         ^~~~~~~~~~
  cargo:warning=1 error generated.
  exit status: 1

  --- stderr

  error occurred: Command "cc" "-O3" "-ffunction-sections" "-fdata-sections" "-fPIC" "-m64" "-arch" "x86_64" "-I" "/var/folders/4y/h7ws2sw13n36hvn3x20c89hm0000gn/T/cargo-installL8TD6a/release/build/libusb1-sys-892fc22518a7ef9c/out/include" "-I" "libusb/libusb" "-Wall" "-Wextra" "-DPRINTF_FORMAT(a, b)=" "-DENABLE_LOGGING=1" "-DOS_DARWIN=1" "-DTARGET_OS_OSX=1" "-DHAVE_SYS_TIME_H=1" "-DHAVE_NFDS_T=1" "-DPLATFORM_POSIX=1" "-DHAVE_CLOCK_GETTIME=1" "-DDEFAULT_VISIBILITY=__attribute__((visibility(\"default\")))" "-o" "/var/folders/4y/h7ws2sw13n36hvn3x20c89hm0000gn/T/cargo-installL8TD6a/release/build/libusb1-sys-892fc22518a7ef9c/out/libusb/libusb/os/darwin_usb.o" "-c" "libusb/libusb/os/darwin_usb.c" with args "cc" did not execute successfully (status code exit status: 1).

warning: build failed, waiting for other jobs to finish...
error: failed to compile `wchisp v0.1.3 (https://github.com/ch32-rs/wchisp#aedf69aa)`, intermediate artifacts can be found at `/var/folders/4y/h7ws2sw13n36hvn3x20c89hm0000gn/T/cargo-installL8TD6a`

andelf, 2022-08-10:
This is a dependency build error. Try use `brew install libusb`.

wankey, 2022-08-10:
> This is a dependency build error. Try use `brew install libusb`.

I have install it already.

andelf, 2022-08-10:
I checked https://github.com/a1ien/rusb/blob/master/libusb1-sys/README.md
It says it also requires `pkc-config`, mine is:

```console
$ which pkg-config
/opt/homebrew/bin/pkg-config
```

wankey, 2022-08-10:
pkg-config is also installed. 
it seems pkg-config cannot find libusb

wankey, 2022-08-10:
> pkg-config is also installed. it seems pkg-config cannot find libusb

UPDATE: 
after add libusb into PKG_CONFIG_PATH, it still build failed.

I tried clone the libusb1-sys and run cargo build successfully.

So, maybe other problem occurred?

wankey, 2022-08-10:
After a day of searching on Google, installed it successfully.

here are some possible solution:
1. add libusb to PKG_CONFIG_PATH(export PKG_CONFIG_PATH="/usr/local/opt/libusb/lib/pkgconfig:${PKG_CONFIG_PATH}")
2. export SDKROOT=$(xcrun --show-sdk-path --sdk macosx)

andelf, 2022-08-10:
I think the `pkg-config` doc for adding environment variables only shows once after the first installation.🤣

closed 2022-08-11 by wankey