ch32-rs/wchisp #10: Fail to flash CH552 because of isp_key checksum mismatch

elfmimi, 2022-08-10:
Thanks for sharing this promising work of yours.

In short, with CH551/CH552/CH554/CH559, checksum have to be calculated from the first 4 bytes of chip's UID.

it seem 'device_type == 0x11' is the condition for when it should be handled specially.

and, for just in case, modified calculation will look like this.
```
        let checksum = self
            .chip_uid[0..4]
            .iter()
            .fold(0_u8, |acc, &x| acc.wrapping_add(x));
```

I only tested with CH552 , BTVER = v2.4.0 .

andelf, 2022-08-10:
Thanks for your report. I have a CH559. Will check this.

elfmimi, 2022-08-30:
I tested another CH552 which has bootloader v2.31 .
This one reports its Chip UID as 7e-4a-45-43-77-7f-af-fb .
Luckily or unluckily isp_key's checksum matches whether using 4 bytes or 8 bytes for calculation.
Using all 8 bytes of UID , it will succeed to flash but it won't run correctly. because of wrongly calculated isp_key .
Using first 4 bytes of UID , it will also succeed to flash and in this case it will run correctly.

elfmimi, 2022-08-30:
Looking at https://github.com/ch32-rs/wchisp/issues/14 ,
it seems bootloader v2.50 decided to report UID with its latter part filled with zeros.
so that isp_key won't get wrongly calculated.