ch32-rs/wchisp #12: ch582: wchisp eeprom hangs

karlp, 2022-08-12:
(presumably crashes the bootloader too, no other commands work until it's power cycled
```
karlp@beros:~/src/wch-ch552tool (master)$ wchisp info
22:11:28 [INFO] Found USB Device Bus 001 Device 038: ID 4348:55e0
22:11:28 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
22:11:28 [DEBUG] (1) wchisp::transport: <= a1000200 8216
22:11:28 [DEBUG] (1) wchisp::flashing: found chip: CH582[0x8216]
22:11:28 [DEBUG] (1) wchisp::transport: => a70200   1f00
22:11:28 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00ffffffff23000000d50fff4f000204000d1c34263b387c7a
22:11:28 [DEBUG] (1) wchisp::flashing: read_config: ffffffff23000000d50fff4f000204000d1c34263b387c7a
22:11:28 [INFO] Chip: CH582[0x8216] (Code Flash: 448KiB, Data EEPROM: 32KiB)
22:11:28 [INFO] Chip UID: 0d-1c-34-26-3b-38-7c-7a
22:11:28 [INFO] BTVER(bootloader ver): 02.40
22:11:28 [DEBUG] (1) wchisp::transport: => a70200   0700
22:11:28 [DEBUG] (1) wchisp::transport: <= a7000e00 0700ffffffff23000000d50fff4f
RESERVED: 0xFFFFFFFF
WPROTECT: 0x00000023
  [0:0] NO_KEY_SERIAL_DOWNLOAD 0b1 (0x1)
    `- Enable
  [1:1] DOWNLOAD_CFG 0b1 (0x1)
    `- PB22(Default set)
CONFIG: 0x4FFF0FD5
  [2:0] RESERVED 0b101 (0x5)
  [3:3] CFG_RESET_EN 0b0 (0x0)
    `- Disable
  [4:4] CFG_DEBUG_EN 0b1 (0x1)
    `- Enable
  [5:5] RESERVED 0b0 (0x0)
  [6:6] CFG_BOOT_EN 0b1 (0x1)
    `- Enable
  [7:7] CFG_ROM_READ 0b1 (0x1)
    `- Read enable
  [27:8] RESERVED 0b11111111111100001111 (0xFFF0F)
  [31:28] VALID_SIG 0b100 (0x4)
    `- Valid
karlp@beros:~/src/wch-ch552tool (master)$ wchisp eeprom
22:11:44 [INFO] Found USB Device Bus 001 Device 038: ID 4348:55e0
22:11:44 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e

^C
karlp@beros:~/src/wch-ch552tool (master)$ wchisp config
22:12:17 [INFO] Found USB Device Bus 001 Device 038: ID 4348:55e0
22:12:17 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
c^C
karlp@beros:~/src/wch-ch552tool (master)$ wchisp info
22:12:21 [INFO] Found USB Device Bus 001 Device 038: ID 4348:55e0
22:12:21 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
Error: Pipe error

```

karlp, 2022-08-12:
Flashing fails too, at least on my board...

```
(wchisp-info as above, freshly replugged then...)
karlp@beros:~/src/wch-ch552tool (master)$ wchisp flash /home/karlp/src/wch-ch583.git/EVT/EXAM/UART1/lolicats.bin
22:18:00 [INFO] Found USB Device Bus 001 Device 040: ID 4348:55e0
22:18:00 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
times out here...
```

andelf, 2022-11-13:
Fixed by increasing USB packet timeout.

closed 2022-11-13 by andelf

andelf, 2022-11-13:
```console
> cargo run -- eeprom ./data.bin
....
> hexdump -C data.bin
00000000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  ||
*
00000020  ff 32 33 34 32 33 35 31  33 35 34 31 32 33 34 ff  |23423513541234|
00000030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  ||
*
00000080  ff ff ff ff 61 73 64 66  61 73 64 66 61 73 64 66  |asdfasdfasdf|
00000090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  ||
*
00008000
```

commit 3e1ad0fb (2022-11-13): feat: impl eeprom dump

Close #12