ch32-rs/wchisp #18: WCH579M info ends abruptly (?)

Nexxkinn, 2022-11-12:
```
09:38:48 [INFO] Found USB Device Bus 001 Device 018: ID 4348:55e0
09:38:48 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
09:38:48 [DEBUG] (1) wchisp::transport: <= a1000200 7913
09:38:48 [DEBUG] (1) wchisp::flashing: found chip: CH579[0x7913]
09:38:48 [DEBUG] (1) wchisp::transport: => a70200   1f00
09:38:48 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00ffffffffffffffff48ffff5000020900f60826263b385767
09:38:48 [DEBUG] (1) wchisp::flashing: read_config: ffffffffffffffff48ffff5000020900f60826263b385767
09:38:48 [INFO] Chip: CH579[0x7913] (Code Flash: 250KiB, Data EEPROM: 2KiB)
09:38:48 [INFO] Chip UID: f6-08-26-26-3b-38-57-67
09:38:48 [INFO] BTVER(bootloader ver): 02.90
09:38:48 [DEBUG] (1) wchisp::transport: => a70200   0700
09:38:48 [DEBUG] (1) wchisp::transport: <= a7000e00 0700ffffffffffffffff48ffff50
```

Attempting to rerun the tool again give me `Error: Input/Output Error` instead. 

Any Idea why?

andelf, 2022-11-13:
Sorry for the inconvenience. The config register info is not filled yet for the CH57x family(requires some USB suffering and matching against the register map document). 
I do not have a CH579 board for now.🥲

Nexxkinn, 2022-11-13:
Bummer 🤪

Please let me know if you're adding support for CH57x family :+1: 

Here's the store I bought from, for reference.

https://aliexpress.com/item/1005003651738891.html

andelf, 2022-11-17:
@Nexxkinn I've just added support for CH579. Could you have a try?

Nexxkinn, 2022-11-19:
@andelf 

`wchisp info`
```
09:35:55 [INFO] Chip: CH579[0x7913] (Code Flash: 250KiB, Data EEPROM: 2KiB)
09:35:55 [INFO] Chip UID: f6-08-26-26-3b-38-57-67
09:35:55 [INFO] BTVER(bootloader ver): 02.90
09:35:55 [INFO] Current config registers: ffffffffffffffff4fffff50
RESERVED: 0xFFFFFFFF
WPROTECT: 0xFFFFFFFF
  [0:0]   NO_KEY_SERIAL_DOWNLOAD 0x1 (0b1)
    `- Enable
  [1:1]   DOWNLOAD_CFG 0x1 (0b1)
    `- PB22(Default set)
USER_CFG: 0x50FFFF4F
  [2:0]   RESERVED 0x7 (0b111)
    `- Error
  [3:3]   CFG_RESET_EN 0x1 (0b1)
    `- Enable
  [4:4]   CFG_DEBUG_EN 0x0 (0b0)
    `- Disable
  [5:5]   RESERVED 0x0 (0b0)
    `- Default
  [6:6]   CFG_BOOT_EN 0x1 (0b1)
    `- Enable
  [7:7]   CFG_ROM_READ 0x0 (0b0)
    `- Disable the programmer to read out, and keep the program secret
  [27:8]  RESERVED 0xFFFF (0b1111111111111111)
    `- Default
  [31:28] VALID_SIG 0x5 (0b101)
    `- Valid
```

`wchisp flash`
```
09:38:52 [INFO] Chip: CH579[0x7913] (Code Flash: 250KiB, Data EEPROM: 2KiB)
09:38:52 [INFO] Chip UID: f6-08-26-26-3b-38-57-67
09:38:52 [INFO] BTVER(bootloader ver): 02.90
09:38:52 [INFO] Current config registers: ffffffffffffffff4fffff50
RESERVED: 0xFFFFFFFF
WPROTECT: 0xFFFFFFFF
  [0:0]   NO_KEY_SERIAL_DOWNLOAD 0x1 (0b1)
    `- Enable
  [1:1]   DOWNLOAD_CFG 0x1 (0b1)
    `- PB22(Default set)
USER_CFG: 0x50FFFF4F
  [2:0]   RESERVED 0x7 (0b111)
    `- Error
  [3:3]   CFG_RESET_EN 0x1 (0b1)
    `- Enable
  [4:4]   CFG_DEBUG_EN 0x0 (0b0)
    `- Disable
  [5:5]   RESERVED 0x0 (0b0)
    `- Default
  [6:6]   CFG_BOOT_EN 0x1 (0b1)
    `- Enable
  [7:7]   CFG_ROM_READ 0x0 (0b0)
    `- Disable the programmer to read out, and keep the program secret
  [27:8]  RESERVED 0xFFFF (0b1111111111111111)
    `- Default
  [31:28] VALID_SIG 0x5 (0b101)
    `- Valid
09:38:52 [INFO] Read embedded_playground/ch-blink/target/thumbv6m-none-eabi/release/ch-blink as ELF format
09:38:52 [INFO] Found loadable segment, physical address: 0x00000000, virtual address: 0x00000000, flags: 0x4
09:38:52 [INFO] Section names: [".vector_table"]
09:38:52 [INFO] Found loadable segment, physical address: 0x00000090, virtual address: 0x00000090, flags: 0x5
09:38:52 [INFO] Section names: [".text"]
09:38:52 [INFO] Firmware size: 1024
09:38:52 [INFO] Erasing...
09:38:52 [WARN] erase_code: set min number of erased sectors to 8
09:38:53 [INFO] Erased 8 code flash sectors
09:38:54 [INFO] Erase done
09:38:54 [INFO] Writing to code flash...
███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████ 1024/102409:38:54 [INFO] Code flash 1024 bytes written
09:38:55 [INFO] Verifying...
███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████ 1024/102409:38:55 [INFO] Verify OK
09:38:55 [INFO] Now reset device and skip any communication errors
09:38:55 [INFO] Device reset
```

The device now properly run a simple code to turn on PB0/LED2 :+1: 

[image]

andelf, 2022-11-19:
@Nexxkinn Thanks.
EEPROM(Data flash) erase/write/dump should also work.

Nexxkinn, 2022-11-20:
> @Nexxkinn Thanks. EEPROM(Data flash) erase/write/dump should also work.

Dumping and writing the eeprom has been confirmed to work on my device too.

Closing the issue, as the tool works generally fine for now.

closed 2022-11-20 by Nexxkinn