ch32-rs/wchisp #20: Windows CH583 Error: Operation not supported or unimplemented on this platform

sabas1080, 2022-12-06:
Hi

I have the next error with CH583 and Windows 

`Error: Operation not supported or unimplemented on this platform`

andelf, 2022-12-06:
You might need to install libusb/winusb driver on Windows.
See-also: https://zadig.akeo.ie/

rgoulter, 2023-11-13:
I've been trying this tool with the CH552 (bootloader version 02.50), on Linux and on Windows.

It works as expected on Linux.

On Windows, my experience has been much flakier.

Sometimes, the bootloader device is able to connect (& firmware can be flashed onto it); most of the time, the device does not connect/start.

The problems are unrelated to this codebase.

With the driver libusb-win32, I did see "Error: Input/Output Error" when trying `wchisp probe`.

With the WinUSB or libusbK drivers (installed using Zadig), ~95% of the time, there was some problem when trying to connect to the bootloader. (In device manager, the bootloader device has a yellow warning icon, and has the status "This device cannot start. (Code 10)"). Of course, if the device does not successfully connected, then there will be an error for `wchisp probe`.

The times where the bootloader was able to connect, I was able to use `wchisp`'s `probe`, `info`, and `flash` just fine.

I'm not sure why it works better on Linux than on Windows.