ch32-rs/wchisp #21: WCH-LinkE-R0-1V2 usb probe to program CH32V003

Martoni, 2022-12-11:
Is wchisp usable for[ WCH-LinkE-R0-1V2](https://www.tindie.com/products/adz1122/wch-linke-online-download-debugger-ch32v003/) usb probe bought on tindie. To program [CH32V003](https://www.tindie.com/products/adz1122/ch32v003-risc-v-mcu-development-board/) devboard  ?

I tried it with mine but I get USB error : 

```shell
$ dmesg | tail
[43923.137566] usb 1-1.2.4: new full-speed USB device number 89 using ehci-pci
[43923.250687] usb 1-1.2.4: New USB device found, idVendor=1a86, idProduct=8010, bcdDevice= 2.07
[43923.250691] usb 1-1.2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[43923.250693] usb 1-1.2.4: Product: WCH-Link
[43923.250694] usb 1-1.2.4: Manufacturer: wch.cn
[43923.250695] usb 1-1.2.4: SerialNumber: 0001A0000000
[43923.251209] cdc_acm 1-1.2.4:1.1: ttyACM0: USB ACM device
[44541.144598] usb 1-1.2.4: USB disconnect, device number 89
[44543.414701] usb 1-1.2.4: new full-speed USB device number 90 using ehci-pci
[44543.523873] usb 1-1.2.4: New USB device found, idVendor=1a86, idProduct=8010, bcdDevice= 2.07
[44543.523877] usb 1-1.2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[44543.523879] usb 1-1.2.4: Product: WCH-Link
[44543.523880] usb 1-1.2.4: Manufacturer: wch.cn
[44543.523881] usb 1-1.2.4: SerialNumber: 0001A0000000
[44543.524408] cdc_acm 1-1.2.4:1.1: ttyACM0: USB ACM device
$ wchisp info
Error: No WCH ISP USB device found(4348:55e0 device not found at index #0)
```

andelf, 2022-12-12:
wchisp is design to use MCU's native USB ISP protocol.
For WCH-Link, please use openocd.

Martoni, 2022-12-12:
Ok, thank you for this quick response.

closed 2022-12-12 by Martoni