ch32-rs/wchisp #23: works fine on 582F from weact

alinelena, 2023-02-12:
```
wchisp config info                                                                                                                                                                   1 ⨯ 12/02/23 09:47:19
09:47:27 [INFO] Current config registers: ffffffffffffffff450fff4f
RESERVED: 0xFFFFFFFF
WPROTECT: 0xFFFFFFFF
  [0:0]   NO_KEY_SERIAL_DOWNLOAD 0x1 (0b1)
    `- Enable
  [1:1]   DOWNLOAD_CFG 0x1 (0b1)
    `- PB22(Default set)
USER_CFG: 0x4FFF0F45
  [2:0]   RESERVED 0x5 (0b101)
  [3:3]   CFG_RESET_EN 0x0 (0b0)
    `- Disable
  [4:4]   CFG_DEBUG_EN 0x0 (0b0)
    `- Disable
  [5:5]   RESERVED 0x0 (0b0)
  [6:6]   CFG_BOOT_EN 0x1 (0b1)
    `- Enable
  [7:7]   CFG_ROM_READ 0x0 (0b0)
    `- Disable the programmer to read out, and keep the program secret
  [27:8]  RESERVED 0xFFF0F (0b11111111111100001111)
  [31:28] VALID_SIG 0x4 (0b100)
    `- Valid
```

no idea what is the bootloader version

andelf, 2023-02-12:
What is the version? It should be something like:

```
15:29:49 [INFO] Chip: CH32V203G6U6[0x3619] (Code Flash: 32KiB)
15:29:49 [INFO] Chip UID: cd-ab-b4-ae-45-bc-c6-16
15:29:49 [INFO] BTVER(bootloader ver): 02.60
15:29:49 [INFO] Code Flash protected: false
15:29:49 [INFO] Current config registers: a55aff0000ff00ffffffffff
```

alinelena, 2023-02-12:
sorry did not pay attention during flash

```
16:21:05 [INFO] Chip: CH582[0x8216] (Code Flash: 448KiB, Data EEPROM: 32KiB)
16:21:05 [INFO] Chip UID: AF-0D-D5-26-3B-38-BF-6C
16:21:05 [INFO] BTVER(bootloader ver): 02.40
16:21:05 [INFO] Current config registers: ffffffffffffffff450fff4f
```

closed 2024-10-01 by andelf