ch32-rs/wchisp #24: Windows unsupported?

maxgerhardt, 2023-02-19:
Using the latest binaries from the CI for Windows
```
C:\Users\Max>C:\Users\Max\Downloads\wchisp.exe probe
19:15:25 [INFO] Found 1 devices
19:15:25 [INFO] hint: use `wchisp info` to check chip info
Error: Operation not supported or unimplemented on this platform

C:\Users\Max>C:\Users\Max\Downloads\wchisp.exe info
Error: Operation not supported or unimplemented on this platform
```

Using the latest binaries from the CI in Ubuntu 22.04

```
max@virtualbox:~/ch32$ sudo ./wchisp probe
19:22:18 [INFO] Found 1 devices
19:22:18 [INFO] hint: use `wchisp info` to check chip info
Device #0: CH32V307VCT6[0x7017]
```

Why are there binaries for Windows available when it can't do anything? 😅

maxgerhardt, 2023-02-19:
Thanks to having MounRiver Studio installed previously, the drivers I'm getting are the wch.cn ones.

[image]

Does this expect WinUSB or LibUSB and can't work with the official WCH.cn drivers?

andelf, 2023-02-20:
Yes. libusb requires the libusb/WinUSB drivers to work.

andelf, 2023-02-20:
Added this note to README.

maxgerhardt, 2023-02-20:
Can we modify the program to print a more helpful error message if this error occurs? "Likely no WinUSB / LibUSB drivers installed, please use Zadig.."

andelf, 2023-02-20:
Good point. Added in be71a08

maxgerhardt, 2023-02-20:
Great! I'll retry on Windows + Linux later today to make sure the error messages trigger and communication works with WinUSB/LibUSB drivers, then the issue can be closed.

maxgerhardt, 2023-02-20:
Works on WinUSB drivers.
```
C:\Users\Max\temp\wchisp>C:\Users\Max\Downloads\wchisp.exe info
21:19:12 [INFO] Chip: CH32V307VCT6[0x7017] (Code Flash: 256KiB)
21:19:12 [INFO] Chip UID: 90-D7-3E-26-3B-38-09-36
21:19:12 [INFO] BTVER(bootloader ver): 02.80
21:19:12 [INFO] Code Flash protected: true
21:19:12 [INFO] Current config registers: ff003fc000ff00ffffffffff
RDPR_USER: 0xC03F00FF
  [7:0]   RDPR 0xFF (0b11111111)
    `- Protected
  [16:16] IWDG_SW 0x1 (0b1)
    `- IWDG enabled by the software, and disabled by hardware
  [17:17] STOP_RST 0x1 (0b1)
    `- Disable
  [18:18] STANDBY_RST 0x1 (0b1)
    `- Disable, entering standby-mode without RST
  [23:22] SRAM_CODE_MODE 0x0 (0b0)
    `- CODE-192KB + RAM-128KB
DATA: 0xFF00FF00
  [7:0]   DATA0 0x0 (0b0)
  [23:16] DATA1 0x0 (0b0)
WRP: 0xFFFFFFFF
  `- Unprotected
```

Without them I get the new error message
```
C:\Users\Max\temp\wchisp>C:\Users\Max\Downloads\wchisp.exe info
21:00:51 [ERROR] Failed to open USB device: Bus 002 Device 002: ID 4348:55e0
21:00:51 [WARN] It's likely no WinUSB/LibUSB drivers insalled. Please install it from Zadig. See also: https://zadig.akeo.ie
Error: Failed to open USB device on Windows
```

Note: The CI action for the Rust build has not triggered for the latest code changes, I had to fork it and trigger it myself.

closed 2023-02-20 by maxgerhardt