ch32-rs/wchisp #25: Arch Linux build errors

taotieren, 2023-02-20:
[AUR wchisp-git](https://aur.archlinux.org/packages/wchisp-git)

```bash
❯ makepkg -sf
==> Making package: wchisp-git 0.2.0.r5.gb3143d6-1 (Mon 20 Feb 2023 10:13:44 AM CST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating wchisp git repo...
remote: Enumerating objects: 106, done.
remote: Counting objects: 100% (106/106), done.
remote: Compressing objects: 100% (66/66), done.
remote: Total 106 (delta 67), reused 79 (delta 40), pack-reused 0
Receiving objects: 100% (106/106), 26.12 KiB | 361.00 KiB/s, done.
Resolving deltas: 100% (67/67), completed with 6 local objects.
From https://github.com/ch32-rs/wchisp
   b3143d6..c03033c  main       -> main
 * [new tag]         v0.2.2     -> v0.2.2
==> Validating source files with sha256sums...
    wchisp ... Skipped
==> Extracting sources...
  -> Creating working copy of wchisp git repo...
Cloning into 'wchisp'...
done.
==> Starting pkgver()...
==> Updated version: wchisp-git 0.2.2.r1.gc03033c-1
==> Removing existing $pkgdir/ directory...
==> Starting build()...
    Updating `rsproxy` index
   Compiling libc v0.2.139
   Compiling proc-macro2 v1.0.51
   Compiling unicode-ident v1.0.6
   Compiling quote v1.0.23
   Compiling syn v1.0.107
   Compiling version_check v0.9.4
   Compiling cfg-if v1.0.0
   Compiling io-lifetimes v1.0.5
   Compiling pkg-config v0.3.26
   Compiling cc v1.0.79
   Compiling autocfg v1.1.0
   Compiling rustix v0.36.8
   Compiling serde_derive v1.0.152
   Compiling proc-macro-error-attr v1.0.4
   Compiling proc-macro-error v1.0.4
   Compiling getrandom v0.2.8
   Compiling indexmap v1.9.2
   Compiling time-core v0.1.0
   Compiling itoa v1.0.5
   Compiling memchr v2.5.0
   Compiling serde v1.0.152
   Compiling log v0.4.17
   Compiling bitflags v1.3.2
   Compiling linux-raw-sys v0.1.4
   Compiling portable-atomic v0.3.19
   Compiling time-macros v0.2.7
   Compiling rand_core v0.6.4
   Compiling heck v0.4.1
   Compiling ppv-lite86 v0.2.17
   Compiling unicode-width v0.1.10
   Compiling lazy_static v1.4.0
   Compiling termcolor v1.1.3
   Compiling rusb v0.9.1
   Compiling hashbrown v0.12.3
   Compiling anyhow v1.0.69
   Compiling os_str_bytes v6.4.1
   Compiling num_threads v0.1.6
   Compiling clap_lex v0.3.1
   Compiling is-terminal v0.4.3
   Compiling console v0.15.5
   Compiling rand_chacha v0.3.1
[... 73 lines cut ...]
          /usr/bin/ld: /home/taotieren/git_clone/aur/wchisp-git/src/wchisp/target/release/deps/libwchisp-52de9e0949f2101c.rlib(wchisp-52de9e0949f2101c.wchisp.9674db75-cgu.6.rcgu.o): in function `rusb::device::Device<T>::active_config_descriptor':
          wchisp.9674db75-cgu.6:(.text._ZN4rusb6device15Device$LT$T$GT$24active_config_descriptor17h3e5563cf077a03bfE+0x13): undefined reference to `libusb_get_active_config_descriptor'
          /usr/bin/ld: /home/taotieren/git_clone/aur/wchisp-git/src/wchisp/target/release/deps/libwchisp-52de9e0949f2101c.rlib(wchisp-52de9e0949f2101c.wchisp.9674db75-cgu.6.rcgu.o): in function `<rusb::device::Device<T> as core::fmt::Debug>::fmt':
          wchisp.9674db75-cgu.6:(.text._ZN66_$LT$rusb..device..Device$LT$T$GT$$u20$as$u20$core..fmt..Debug$GT$3fmt17h80696cd3aa3adea8E+0x22): undefined reference to `libusb_get_device_descriptor'
          /usr/bin/ld: wchisp.9674db75-cgu.6:(.text._ZN66_$LT$rusb..device..Device$LT$T$GT$$u20$as$u20$core..fmt..Debug$GT$3fmt17h80696cd3aa3adea8E+0xb6): undefined reference to `libusb_get_bus_number'
          /usr/bin/ld: wchisp.9674db75-cgu.6:(.text._ZN66_$LT$rusb..device..Device$LT$T$GT$$u20$as$u20$core..fmt..Debug$GT$3fmt17h80696cd3aa3adea8E+0xc3): undefined reference to `libusb_get_device_address'
          /usr/bin/ld: /home/taotieren/git_clone/aur/wchisp-git/src/wchisp/target/release/deps/librusb-4c2828f903d02b30.rlib(rusb-4c2828f903d02b30.rusb.f34616fc-cgu.3.rcgu.o): in function `<rusb::config_descriptor::ConfigDescriptor as core::ops::drop::Drop>::drop':
          rusb.f34616fc-cgu.3:(.text._ZN83_$LT$rusb..config_descriptor..ConfigDescriptor$u20$as$u20$core..ops..drop..Drop$GT$4drop17hefcb2b5a2d9a27bcE+0x5): undefined reference to `libusb_free_config_descriptor'
          /usr/bin/ld: /home/taotieren/git_clone/aur/wchisp-git/src/wchisp/target/release/deps/librusb-4c2828f903d02b30.rlib(rusb-4c2828f903d02b30.rusb.f34616fc-cgu.4.rcgu.o): in function `core::ptr::drop_in_place<alloc::sync::ArcInner<rusb::context::ContextInner>>':
          rusb.f34616fc-cgu.4:(.text._ZN4core3ptr77drop_in_place$LT$alloc..sync..ArcInner$LT$rusb..context..ContextInner$GT$$GT$17h5afaeeb33e032a52E+0x2): undefined reference to `libusb_exit'
          /usr/bin/ld: /home/taotieren/git_clone/aur/wchisp-git/src/wchisp/target/release/deps/librusb-4c2828f903d02b30.rlib(rusb-4c2828f903d02b30.rusb.f34616fc-cgu.4.rcgu.o): in function `<rusb::context::ContextInner as core::ops::drop::Drop>::drop':
          rusb.f34616fc-cgu.4:(.text._ZN69_$LT$rusb..context..ContextInner$u20$as$u20$core..ops..drop..Drop$GT$4drop17h5a04276ff2486cb4E+0x5): undefined reference to `libusb_exit'
          /usr/bin/ld: /home/taotieren/git_clone/aur/wchisp-git/src/wchisp/target/release/deps/librusb-4c2828f903d02b30.rlib(rusb-4c2828f903d02b30.rusb.f34616fc-cgu.4.rcgu.o): in function `rusb::context::Context::new':
          rusb.f34616fc-cgu.4:(.text._ZN4rusb7context7Context3new17h9aef9d72f8044560E+0xc): undefined reference to `libusb_init'
          collect2: error: ld returned 1 exit status
          
  = note: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#cargorustc-link-libkindname)

error: could not compile `wchisp` due to previous error
==> ERROR: A failure occurred in build().
    Aborting...

```

andelf, 2023-02-20:
libusb is required to build wchisp.

syx-413, 2023-12-27:
https://archlinux.org/packages/?sort=&q=libusb&maintainer=&flagged=

closed 2024-10-03 by taotieren