ch32-rs/wchisp #26: wchisp flash is not work as expected

hc7, 2023-02-22:
Hello, please explain why I can't flash wch583:

:~/work/b9/debug$ wchisp info
09:52:44 [INFO] Chip: CH582[0x8216] (Code Flash: 448KiB, Data EEPROM: 32KiB)
09:52:44 [INFO] Chip UID: A7-06-86-26-3B-38-68-65
09:52:44 [INFO] BTVER(bootloader ver): 02.40
09:52:44 [INFO] Current config registers: ffffffffffffffffd50fff4f
RESERVED: 0xFFFFFFFF
WPROTECT: 0xFFFFFFFF
  [0:0]   NO_KEY_SERIAL_DOWNLOAD 0x1 (0b1)
    `- Enable
  [1:1]   DOWNLOAD_CFG 0x1 (0b1)
    `- PB22(Default set)
USER_CFG: 0x4FFF0FD5
  [2:0]   RESERVED 0x5 (0b101)
  [3:3]   CFG_RESET_EN 0x0 (0b0)
    `- Disable
  [4:4]   CFG_DEBUG_EN 0x1 (0b1)
    `- Enable
  [5:5]   RESERVED 0x0 (0b0)
  [6:6]   CFG_BOOT_EN 0x1 (0b1)
    `- Enable
  [7:7]   CFG_ROM_READ 0x1 (0b1)
    `- Read enable
  [27:8]  RESERVED 0xFFF0F (0b11111111111100001111)
  [31:28] VALID_SIG 0x4 (0b100)
    `- Valid
:~/work/b9/debug$ 

Is there wome way to understand what problem here?
Log:
:~/work/b9/debug$ wchisp flash setup.hex 
09:53:08 [INFO] Chip: CH582[0x8216] (Code Flash: 448KiB, Data EEPROM: 32KiB)
09:53:08 [INFO] Chip UID: A7-06-86-26-3B-38-68-65
09:53:08 [INFO] BTVER(bootloader ver): 02.40
09:53:08 [INFO] Current config registers: ffffffffffffffffd50fff4f
RESERVED: 0xFFFFFFFF
WPROTECT: 0xFFFFFFFF
  [0:0]   NO_KEY_SERIAL_DOWNLOAD 0x1 (0b1)
    `- Enable
  [1:1]   DOWNLOAD_CFG 0x1 (0b1)
    `- PB22(Default set)
USER_CFG: 0x4FFF0FD5
  [2:0]   RESERVED 0x5 (0b101)
  [3:3]   CFG_RESET_EN 0x0 (0b0)
    `- Disable
  [4:4]   CFG_DEBUG_EN 0x1 (0b1)
    `- Enable
  [5:5]   RESERVED 0x0 (0b0)
  [6:6]   CFG_BOOT_EN 0x1 (0b1)
    `- Enable
  [7:7]   CFG_ROM_READ 0x1 (0b1)
    `- Read enable
  [27:8]  RESERVED 0xFFF0F (0b11111111111100001111)
  [31:28] VALID_SIG 0x4 (0b100)
    `- Valid
09:53:08 [INFO] Read setup.hex as IntelHex format
09:53:08 [INFO] Firmware size: 439296
09:53:08 [INFO] Erasing...
09:53:08 [INFO] Erased 430 code flash sectors
09:53:09 [INFO] Erase done
09:53:09 [INFO] Writing to code flash...
██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████ 439296/43929609:53:11 [INFO] Code flash 439296 bytes written
09:53:11 [INFO] Verifying...
Error: Verify failed, mismatch

andelf, 2023-02-22:
I can reproduce this too.
This is a regression. 🥲

andelf, 2023-02-22:
@hc7 I used wchisp to successfully flash my CH582. So this must be some bug introduced recently.

After I downloaded the windows version of WCHISPTool.
The WCHISPTool fails too!! 

Then I realized, this is caused by CFG flags.
When CFG flag is `4d0fff4f`, then you can flash the MCU with wchisp.
When CFG flag is `4FFF0FD5`, then the verify process always fails. 🤯

I changed default reset value of CFG flag recently to help developing a wch-link tool.

andelf, 2023-02-22:
```consle
> wchisp config reset
17:16:22 [INFO] Current config registers: ffffffffffffffffdf3f0f4d
17:16:22 [INFO] Reset config registers:   ffffffffffffffff4fff0f4d
> wchisp flash qmk_port_ch582.bin`
17:16:29 [INFO] Chip: CH582[0x8216] (Code Flash: 448KiB, Data EEPROM: 32KiB)
17:16:29 [INFO] Chip UID: E1-75-B5-26-3B-38-D1-D4
17:16:29 [INFO] BTVER(bootloader ver): 02.40
17:16:29 [INFO] Current config registers: ffffffffffffffff4f3f0f4d
RESERVED: 0xFFFFFFFF
WPROTECT: 0xFFFFFFFF
  [0:0]   NO_KEY_SERIAL_DOWNLOAD 0x1 (0b1)
    `- Enable
  [1:1]   DOWNLOAD_CFG 0x1 (0b1)
    `- PB22(Default set)
USER_CFG: 0x4D0F3F4F
  [2:0]   RESERVED 0x7 (0b111)
  [3:3]   CFG_RESET_EN 0x1 (0b1)
    `- Enable
  [4:4]   CFG_DEBUG_EN 0x0 (0b0)
    `- Disable
  [5:5]   RESERVED 0x0 (0b0)
  [6:6]   CFG_BOOT_EN 0x1 (0b1)
    `- Enable
  [7:7]   CFG_ROM_READ 0x0 (0b0)
    `- Disable the programmer to read out, and keep the program secret
  [27:8]  RESERVED 0xD0F3F (0b11010000111100111111)
  [31:28] VALID_SIG 0x4 (0b100)
    `- Valid
17:16:29 [INFO] Read qmk_port_ch582.bin as Binary format
17:16:29 [INFO] Firmware size: 51200
17:16:29 [INFO] Erasing...
17:16:29 [INFO] Erased 51 code flash sectors
17:16:30 [INFO] Erase done
17:16:30 [INFO] Writing to code flash...
██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████ 51200/51200
17:16:32 [INFO] Code flash 51200 bytes written
17:16:32 [INFO] Verifying...
██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████ 51200/51200
17:16:35 [INFO] Verify OK
17:16:35 [INFO] Now reset device and skip any communication errors
17:16:35 [INFO] Device reset
```

hc7, 2023-02-22:
```
:~/work/b9/debug$ wchisp config reset
17:32:19 [INFO] Current config registers: ffffffffffffffffd50fff4f
17:32:19 [INFO] Reset config registers:   ffffffffffffffff4fff0f4d
17:32:19 [INFO] Config register restored to default value(non-protected, debug-enabled)
hcs-i@hcs-i-nix:~/work/b9/debug$ wchisp flash parameters1.hex 
17:32:23 [INFO] Chip: CH582[0x8216] (Code Flash: 448KiB, Data EEPROM: 32KiB)
17:32:23 [INFO] Chip UID: A7-06-86-26-3B-38-68-65
17:32:23 [INFO] BTVER(bootloader ver): 02.40
17:32:23 [INFO] Current config registers: ffffffffffffffff4f3f0f4d
RESERVED: 0xFFFFFFFF
WPROTECT: 0xFFFFFFFF
  [0:0]   NO_KEY_SERIAL_DOWNLOAD 0x1 (0b1)
    `- Enable
  [1:1]   DOWNLOAD_CFG 0x1 (0b1)
    `- PB22(Default set)
USER_CFG: 0x4D0F3F4F
  [2:0]   RESERVED 0x7 (0b111)
  [3:3]   CFG_RESET_EN 0x1 (0b1)
    `- Enable
  [4:4]   CFG_DEBUG_EN 0x0 (0b0)
    `- Disable
  [5:5]   RESERVED 0x0 (0b0)
  [6:6]   CFG_BOOT_EN 0x1 (0b1)
    `- Enable
  [7:7]   CFG_ROM_READ 0x0 (0b0)
    `- Disable the programmer to read out, and keep the program secret
  [27:8]  RESERVED 0xD0F3F (0b11010000111100111111)
  [31:28] VALID_SIG 0x4 (0b100)
    `- Valid
17:32:23 [INFO] Read parameters1.hex as IntelHex format
17:32:23 [INFO] Firmware size: 8192
17:32:23 [INFO] Erasing...
17:32:23 [INFO] Erased 9 code flash sectors
17:32:24 [INFO] Erase done
17:32:24 [INFO] Writing to code flash...
██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████ 8192/819217:32:24 [INFO] Code flash 8192 bytes written
17:32:25 [INFO] Verifying...
██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████ 8192/819217:32:25 [INFO] Verify OK
17:32:25 [INFO] Now reset device and skip any communication errors
17:32:25 [INFO] Device reset
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: NotFound', src/transport/usb.rs:116:36
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
:~/work/b9/debug$ 

```

andelf, 2023-02-22:
That's OK. your devices should be successfully flashed.
The error is caused by an `unwrap()` call in drop(). I'll fix it soon.

hc7, 2023-02-22:
> 

It looks like I haven't flash successfully by wchisp but yes wchisptool can't flash too. But wchisptool can reset bits to working state and then flash.
When enable simulate interface there is no flash.
[image]

When enable code and data protection mode - it is possible to flash and verify.
[image]

andelf, 2023-02-22:
ffffffffffffffff4f3f0f4d should be a valid state.

maxgerhardt, 2023-02-25:
Sadly I'm having the same problem with aa CH32V307, after flashing, verification just fails. Latest commit.

Verbose logs: https://pastebin.com/Mq0WF7Zn

```
\wchisp.exe info
11:41:31 [INFO] Chip: CH32V307VCT6[0x7017] (Code Flash: 256KiB)
11:41:31 [INFO] Chip UID: 90-D7-3E-26-3B-38-09-36
11:41:31 [INFO] BTVER(bootloader ver): 02.80
11:41:31 [INFO] Code Flash protected: true
11:41:31 [INFO] Current config registers: ff003fc000ff00ffffffffff
RDPR_USER: 0xC03F00FF
  [7:0]   RDPR 0xFF (0b11111111)
    `- Protected
  [16:16] IWDG_SW 0x1 (0b1)
    `- IWDG enabled by the software, and disabled by hardware
  [17:17] STOP_RST 0x1 (0b1)
    `- Disable
  [18:18] STANDBY_RST 0x1 (0b1)
    `- Disable, entering standby-mode without RST
  [23:22] SRAM_CODE_MODE 0x0 (0b0)
    `- CODE-192KB + RAM-128KB
DATA: 0xFF00FF00
  [7:0]   DATA0 0x0 (0b0)
  [23:16] DATA1 0x0 (0b0)
WRP: 0xFFFFFFFF
  `- Unprotected
  ```

andelf, 2023-02-25:
@maxgerhardt  The official version has an extra "Deprotect succeed" log.
Could you try `wchisp config reset` then flash?

maxgerhardt, 2023-02-25:
Sadly the same: https://pastebin.com/ubUAUHW6

andelf, 2023-02-25:
🥲 I'll check this following Monday.
Currently, I have no CH32V307 at hand.

andelf, 2023-02-25:
I had some experiments with the WCH-Link protocol at https://github.com/ch32-rs/wchisp

maxgerhardt, 2023-02-25:
One critical thing I noticed in the logs is that although `config reset` was made, the flash command still shows the valeus before the reset. It's as if the reset values weren't saved properly.
```
12:00:17 [INFO] Current config registers: ff003fc000ff00ffffffffff
12:00:17 [INFO] Reset config registers:   a55aff0000ff00ffffffffff
12:00:17 [INFO] Config register restored to default value(non-protected, debug-enabled)

andelf, 2023-02-27:
@maxgerhardt 
Could you try 

```
wchisp -v flash firmware.bin --no-verify
```

Or flash with a large firmware. Then it should back to normal.

It seems the MCU sometimes reports verification results with an 0x0e response.
That happens randomly. 🥲

maxgerhardt, 2023-02-27:
Really weird. I just flashed the MCU again (with verify) and it did not error out in verification. No matter if bigger or smaller firmware. Same size firmware of 9 sectors. Tried at least 20 flashes now and all worked. Hm., I guess I can't complain now.

andelf, 2023-02-28:
Yes. There might be some hidden status.🤣

commit e79cb78f (2023-02-22): fix: reset value of ch58x, fix #26