ch32-rs/wchisp #29: Verify failed for CH32V208WBU6

tonyho, 2023-03-11:
OSInfo:
Description:    Ubuntu 22.04.2 LTS
X64, Linux kernel  5.15.0-57-generic

Use cargo build method or prebuilt ELF from action both have such problem:

```
 wchisp -v flash ../WebServer.hex
02:39:09 [DEBUG] (1) wchisp::transport::usb: Found USB Device Bus 002 Device 006: ID 4348:55e0
02:39:09 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
02:39:09 [DEBUG] (1) wchisp::transport: <= a1000200 8019
02:39:09 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
02:39:09 [DEBUG] (1) wchisp::transport: <= a1000200 8019
02:39:09 [DEBUG] (1) wchisp::flashing: found chip: CH32V208WBU6[0x8019]
02:39:09 [DEBUG] (1) wchisp::transport: => a70200   1f00
02:39:09 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00ff003fc000ff00ffffffffff00020500d04088263b38939f
02:39:09 [DEBUG] (1) wchisp::flashing: read_config: ff003fc000ff00ffffffffff00020500d04088263b38939f
02:39:09 [INFO] Chip: CH32V208WBU6[0x8019] (Code Flash: 128KiB)
02:39:09 [INFO] Chip UID: D0-40-88-26-3B-38-93-9F
02:39:09 [INFO] BTVER(bootloader ver): 02.50
02:39:09 [INFO] Code Flash protected: true
02:39:09 [DEBUG] (1) wchisp::transport: => a70200   0700
02:39:09 [DEBUG] (1) wchisp::transport: <= a7000e00 0700ff003fc000ff00ffffffffff
02:39:09 [INFO] Current config registers: ff003fc000ff00ffffffffff
RDPR_USER: 0xC03F00FF
  [7:0]   RDPR 0xFF (0b11111111)
    `- Protected
  [16:16] IWDG_SW 0x1 (0b1)
    `- IWDG enabled by the software, and disabled by hardware
  [17:17] STOP_RST 0x1 (0b1)
    `- Disable
  [18:18] STANDBY_RST 0x1 (0b1)
    `- Disable, entering standby-mode without RST
  [23:22] SRAM_CODE_MODE 0x0 (0b0)
    `- CODE-192KB + RAM-128KB / CODE-128KB + RAM-64KB depending on the chip
DATA: 0xFF00FF00
  [7:0]   DATA0 0x0 (0b0)
  [23:16] DATA1 0x0 (0b0)
WRP: 0xFFFFFFFF
  `- Unprotected
02:39:09 [INFO] Read ../WebServer.hex as IntelHex format
02:39:09 [INFO] Firmware size: 112640
02:39:09 [INFO] Erasing...
02:39:09 [DEBUG] (1) wchisp::transport: => a40400   6f000000
02:39:09 [DEBUG] (1) wchisp::transport: <= a4000200 0000
02:39:09 [INFO] Erased 111 code flash sectors
02:39:10 [INFO] Erase done
02:39:10 [INFO] Writing to code flash...
02:39:10 [DEBUG] (1) wchisp::transport: => a31e00   000000000000000000000000000000000000000000000000000000000000
02:39:10 [DEBUG] (1) wchisp::transport: <= a3000200 9800
...
02:39:10 [DEBUG] (1) wchisp::transport: <= a5000200 0000
02:39:10 [DEBUG] (1) wchisp::transport: => a51d00   e8b70100b763636363636363e363636363636363e363636363636363e3
02:39:10 [DEBUG] (1) wchisp::transport: <= a5000200 0000
02:39:10 [DEBUG] (1) wchisp::transport: => a50500   00b8010065
02:39:10 [DEBUG] (1) wchisp::transport: <= a5000200 0000
██████████████████████████████████████████████████████████████████████████████
...
112640/11264002:39:10 [INFO] Code flash 112640 bytes written
02:39:11 [INFO] Verifying...
02:39:11 [DEBUG] (1) wchisp::transport: => a31e00   000000000000000000000000000000000000000000000000000000000000
02:39:11 [DEBUG] (1) wchisp::transport: <= a3000200 9800
02:39:11 [DEBUG] (1) wchisp::transport: => a63d00   00000000450c534379706363e370636363706363e370636363706363e370636363706363e370636363706363e370636363706363e370636363106373e3
02:39:11 [DEBUG] (1) wchisp::transport: <= a6000200 f500
Error: Verify failed, mismatch
```

It seems the tool has found the device correctly:

```
$ wchisp -v info
02:44:35 [DEBUG] (1) wchisp::transport::usb: Found USB Device Bus 003 Device 003: ID 4348:55e0
02:44:35 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
02:44:35 [DEBUG] (1) wchisp::transport: <= a1000200 8019
02:44:35 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
02:44:35 [DEBUG] (1) wchisp::transport: <= a1000200 8019
02:44:35 [DEBUG] (1) wchisp::flashing: found chip: CH32V208WBU6[0x8019]
02:44:35 [DEBUG] (1) wchisp::transport: => a70200   1f00
02:44:35 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00ff003fc000ff00ffffffffff00020500d04088263b38939f
02:44:35 [DEBUG] (1) wchisp::flashing: read_config: ff003fc000ff00ffffffffff00020500d04088263b38939f
02:44:35 [INFO] Chip: CH32V208WBU6[0x8019] (Code Flash: 128KiB)
02:44:35 [INFO] Chip UID: D0-40-88-26-3B-38-93-9F
02:44:35 [INFO] BTVER(bootloader ver): 02.50
02:44:35 [INFO] Code Flash protected: true
02:44:35 [DEBUG] (1) wchisp::transport: => a70200   0700
02:44:35 [DEBUG] (1) wchisp::transport: <= a7000e00 0700ff003fc000ff00ffffffffff
02:44:35 [INFO] Current config registers: ff003fc000ff00ffffffffff
RDPR_USER: 0xC03F00FF
  [7:0]   RDPR 0xFF (0b11111111)
    `- Protected
  [16:16] IWDG_SW 0x1 (0b1)
    `- IWDG enabled by the software, and disabled by hardware
  [17:17] STOP_RST 0x1 (0b1)
    `- Disable
  [18:18] STANDBY_RST 0x1 (0b1)
    `- Disable, entering standby-mode without RST
  [23:22] SRAM_CODE_MODE 0x0 (0b0)
    `- CODE-192KB + RAM-128KB / CODE-128KB + RAM-64KB depending on the chip
DATA: 0xFF00FF00
  [7:0]   DATA0 0x0 (0b0)
  [23:16] DATA1 0x0 (0b0)
WRP: 0xFFFFFFFF
  `- Unprotected
```

andelf, 2023-07-02:
I've got a CH32V208WB board.  
Now I can check this.🥲

otipohc69, 2023-07-07:
Doesn't seems to be wchisp error related, your RDPR is protected, just run :
wchisp config unprotect
before trying to flash your board.
Does someone know how to set the registers back for RDPR?