ch32-rs/wchisp #31: Question: Is there a way to programmatically have chips enter bootloader mode?

KurtE, 2023-06-04:
Sorry, not sure if this is the best place to ask this, but wondering if there is a prescribed way for a program that is running on of the chips such as the CH32V307 or CH32V203... to be able to enter into a state where you can use this app to upload the new program. 

I know with some other boards, you can do this by jumping to some special location, others by special instructions, etc.

If there is a way, then I will probably play with the USB code to detect some condition such as some specific set of bytes sent to it, with a specific logical baud rate.  At which point can then potentially build up an external command that can be run within some build process that first uses this to reset the board, and then use wchip to reprogram it.

Thanks and again sorry if this is not the appropriate place to ask this.
Kurt

andelf, 2023-07-09:
I can set PC via [wlink](https://github.com/ch32-rs/wlink), but fail to let it run the bootloader.
Need more investigation.

froscon2024, 2024-10-02:
The installed firmware on the device also make it sort of dependent how you can jump to the bootloader.
For example the closed source firmware on the Badgemagic https://badgemagic.fossasia.org/ did not let jumps to 0x00. This makes it a requirement to solder out the battery once, flash the opensource firmware replacement and then solder on the battery again.
With the opensource firmware now jumping to 0x00 is possible and you do not have to desolder the battery any more: https://github.com/fossasia/badgemagic-firmware/issues/38#issuecomment-2368571276