ch32-rs/wchisp #36: VCRUNTIME140.dll was not found when running prebuilt binary

mejobloggs, 2023-07-10:
I got windows-wchisp-latest from here: `https://github.com/ch32-rs/wchisp/actions/runs/5240843054`

When I try to run it in cmd I get the error `VCRUNTIME140.dll was not found when running prebuilt binary`

Should I be able to run this by itself? I didn't see anything in the instructions about needing to install VCRUNTIME140.dll

Thanks

andelf, 2023-07-10:
Check this: https://learn.microsoft.com/en-us/cpp/windows/latest-supported-vc-redist?view=msvc-170

andelf, 2023-07-10:
This is caused by missing vcredist on your local machine. Rust builds a dynamic linked exe on Windows by default settings.

mejobloggs, 2023-07-10:
Thanks very much. Working after installing that!

Thanks for your work :)

closed 2023-07-10 by mejobloggs