ch32-rs/wchisp #38: Appears to work with CH32X035G8U6

benpye, 2023-08-05:
Poorly documented but the CH32X035 also has a USB bootloader and wchisp seems to work:

```
% sudo ~/.cargo/bin/wchisp --verbose info
22:38:12 [DEBUG] (1) wchisp::transport::usb: Found USB Device Bus 001 Device 001: ID 4348:55e0
22:38:12 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
22:38:12 [DEBUG] (1) wchisp::transport: <= a1000200 5623
22:38:12 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
22:38:12 [DEBUG] (1) wchisp::transport: <= a1000200 5623
22:38:12 [DEBUG] (1) wchisp::flashing: found chip: CH32X035G8U6[0x5623]
22:38:12 [DEBUG] (1) wchisp::transport: => a70200   1f00
22:38:12 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00a55affffffffffffffffffff00020400cdab703848bc85a0
22:38:12 [DEBUG] (1) wchisp::flashing: read_config: a55affffffffffffffffffff00020400cdab703848bc85a0
22:38:12 [INFO] Chip: CH32X035G8U6[0x5623] (Code Flash: 64KiB)
22:38:12 [INFO] Chip UID: CD-AB-70-38-48-BC-85-A0
22:38:12 [INFO] BTVER(bootloader ver): 02.40
22:38:12 [DEBUG] (1) wchisp::transport: => a70200   1f00
22:38:12 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00a55affffffffffffffffffff00020400cdab703848bc85a0
22:38:12 [INFO] Current config registers: a55affffffffffffffffffff00020400cdab703848bc85a0
RDPR_USER: 0xFFFF5AA5
  [7:0]   RDPR 0xA5 (0b10100101)
    `- Unprotected
  [16:16] IWDG_SW 0x1 (0b1)
    `- IWDG enabled by the software, and disabled by hardware
  [17:17] STOP_RST 0x1 (0b1)
    `- Disable
  [18:18] STANDBY_RST 0x1 (0b1)
    `- Disable, entering standby-mode without RST
  [20:19] RST_MOD 0x3 (0b11)
    `- Disable RST alternative function, use PA21/PC3/PB7 as GPIO
DATA: 0xFFFFFFFF
  [7:0]   DATA0 0xFF (0b11111111)
  [23:16] DATA1 0xFF (0b11111111)
WRP: 0xFFFFFFFF
  `- Unprotected
```

andelf, 2023-08-07:
@benpye Thanks for your reminder. I'll update the documents.

froscon2024, 2024-10-02:
@benpye 
Could you tell on what PCB hardware you have tested this to add this hardware design as reported working in the readme?

froscon2024, 2024-10-03:
I understand your plan. 
I created few seconds ago a PR with the current style to add CH549 to the readme here. I would say lets get this merged and additional upcoming ones as long as the PCB-list is missing in https://github.com/ch32-rs/awesome-ch32
I plan to also reorganize the list in the readme to make it sorted alphabetical.

The style i use at the moment is:
[x] Chip name without the additional letter at the end 
 * (manufacturer) (model) (chip name with letter at the end) (Link to picture, schematic or product website)

When you have created the PCB-List in https://github.com/ch32-rs/awesome-ch32 you can move all the lines from this repo here that look like 
 * (manufacturer) (model) (chip name with letter at the end) (Link to picture or schematic)
 and you would be left with the 
[x] Chipname without the additional letter at the end 

And this should be really simple and straight on to move the things over.