ch32-rs/wchisp #43: Part with VID 1a86

dragonlock2, 2024-01-11:
I've come across a part whose bootloader has a VID of `1a86` instead of `4348` which is WCH's other VID. When I modify the `src/transport/usb.rs` to match `1a86` instead, it does show up. Has anyone else seen this?

[image]

```
matthewtran:wchisp $ cargo run -- --verbose info
    Finished dev [unoptimized + debuginfo] target(s) in 0.06s
     Running `target/debug/wchisp --verbose info`
08:39:21 [DEBUG] (1) wchisp::transport::usb: Found USB Device Bus 020 Device 007: ID 1a86:55e0
08:39:21 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
08:39:21 [DEBUG] (1) wchisp::transport: <= a1000200 5623
08:39:21 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
08:39:21 [DEBUG] (1) wchisp::transport: <= a1000200 5623
08:39:21 [DEBUG] (1) wchisp::flashing: found chip: CH32X035G8U6[0x5623]
08:39:21 [DEBUG] (1) wchisp::transport: => a70200   1f00
08:39:21 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00a55a1fe000ff00ffffffffff00020600cdab271649bc3d7e
08:39:21 [DEBUG] (1) wchisp::flashing: read_config: a55a1fe000ff00ffffffffff00020600cdab271649bc3d7e
08:39:21 [INFO] Chip: CH32X035G8U6[0x5623] (Code Flash: 64KiB)
08:39:21 [INFO] Chip UID: CD-AB-27-16-49-BC-3D-7E
08:39:21 [INFO] BTVER(bootloader ver): 02.60
08:39:21 [DEBUG] (1) wchisp::transport: => a70200   1f00
08:39:21 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00a55a1fe000ff00ffffffffff00020600cdab271649bc3d7e
08:39:21 [INFO] Current config registers: a55a1fe000ff00ffffffffff00020600cdab271649bc3d7e
RDPR_USER: 0xE01F5AA5
  [7:0]   RDPR 0xA5 (0b10100101)
    `- Unprotected
  [16:16] IWDG_SW 0x1 (0b1)
    `- IWDG enabled by the software, and disabled by hardware
  [17:17] STOP_RST 0x1 (0b1)
    `- Disable
  [18:18] STANDBY_RST 0x1 (0b1)
    `- Disable, entering standby-mode without RST
  [20:19] RST_MOD 0x3 (0b11)
    `- Disable RST alternative function, use PA21/PC3/PB7 as GPIO
DATA: 0xFF00FF00
  [7:0]   DATA0 0x0 (0b0)
  [23:16] DATA1 0x0 (0b0)
WRP: 0xFFFFFFFF
  `- Unprotected
```

andelf, 2024-01-11:
Thanks. That's interesting. I've ordered some CH32X035 chips to check it.

Nable80, 2024-01-22:
I have the same kind of parts, except that it's ``CH32X035F8U6[0x5e23]`` and it came with a protected test firmware that prints ``this is a test`` into UART2 by default:
```
19:05:45 [DEBUG] (1) wchisp::transport::usb: Found USB Device Bus 003 Device 033: ID 1a86:55e0
19:05:45 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
19:05:45 [DEBUG] (1) wchisp::transport: <= a1000200 5e23
19:05:45 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
19:05:45 [DEBUG] (1) wchisp::transport: <= a1000200 5e23
19:05:45 [DEBUG] (1) wchisp::flashing: found chip: CH32X035F8U6[0x5e23]
19:05:45 [DEBUG] (1) wchisp::transport: => a70200   1f00
19:05:45 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00ff0007f800ff00ffffffffff00020600cdab93c649bca92e
19:05:45 [DEBUG] (1) wchisp::flashing: read_config: ff0007f800ff00ffffffffff00020600cdab93c649bca92e
19:05:45 [INFO] Chip: CH32X035F8U6[0x5e23] (Code Flash: 64KiB)
19:05:45 [INFO] Chip UID: CD-AB-93-C6-49-BC-A9-2E
19:05:45 [INFO] BTVER(bootloader ver): 02.60
19:05:45 [DEBUG] (1) wchisp::transport: => a70200   1f00
19:05:45 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00ff0007f800ff00ffffffffff00020600cdab93c649bca92e
19:05:45 [INFO] Current config registers: ff0007f800ff00ffffffffff00020600cdab93c649bca92e
RDPR_USER: 0xF80700FF
  [7:0]   RDPR 0xFF (0b11111111)
    `- Protected
  [16:16] IWDG_SW 0x1 (0b1)
    `- IWDG enabled by the software, and disabled by hardware
  [17:17] STOP_RST 0x1 (0b1)
    `- Disable
  [18:18] STANDBY_RST 0x1 (0b1)
    `- Disable, entering standby-mode without RST
  [20:19] RST_MOD 0x0 (0b0)
    `- Enable RST alternative function
DATA: 0xFF00FF00
  [7:0]   DATA0 0x0 (0b0)
  [23:16] DATA1 0x0 (0b0)
WRP: 0xFFFFFFFF
  `- Unprotected
```

I erased one chip and it's detected like this:
```
18:59:58 [DEBUG] (1) wchisp::transport::usb: Found USB Device Bus 003 Device 027: ID 1a86:55e0
18:59:58 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
18:59:58 [DEBUG] (1) wchisp::transport: <= a1000200 5e23
18:59:58 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
18:59:58 [DEBUG] (1) wchisp::transport: <= a1000200 5e23
18:59:58 [DEBUG] (1) wchisp::flashing: found chip: CH32X035F8U6[0x5e23]
18:59:58 [DEBUG] (1) wchisp::transport: => a70200   1f00
18:59:58 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00a55affffffffffffffffffff00020600cdab82c649bc982e
18:59:58 [DEBUG] (1) wchisp::flashing: read_config: a55affffffffffffffffffff00020600cdab82c649bc982e
18:59:58 [INFO] Chip: CH32X035F8U6[0x5e23] (Code Flash: 64KiB)
18:59:58 [INFO] Chip UID: CD-AB-82-C6-49-BC-98-2E
18:59:58 [INFO] BTVER(bootloader ver): 02.60
18:59:58 [DEBUG] (1) wchisp::transport: => a70200   1f00
18:59:58 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00a55affffffffffffffffffff00020600cdab82c649bc982e
18:59:58 [INFO] Current config registers: a55affffffffffffffffffff00020600cdab82c649bc982e
RDPR_USER: 0xFFFF5AA5
  [7:0]   RDPR 0xA5 (0b10100101)
    `- Unprotected
  [16:16] IWDG_SW 0x1 (0b1)
    `- IWDG enabled by the software, and disabled by hardware
  [17:17] STOP_RST 0x1 (0b1)
    `- Disable
  [18:18] STANDBY_RST 0x1 (0b1)
    `- Disable, entering standby-mode without RST
  [20:19] RST_MOD 0x3 (0b11)
    `- Disable RST alternative function, use PA21/PC3/PB7 as GPIO
DATA: 0xFFFFFFFF
  [7:0]   DATA0 0xFF (0b11111111)
  [23:16] DATA1 0xFF (0b11111111)
WRP: 0xFFFFFFFF
  `- Unprotected
```

Note: at first your application didn't work for me, I guess it's because the USB bootloader exits after some idle period and I wasn't quick enough.

Nable80, 2024-01-22:
Sometimes I get just this output, which isn't really helpful:
```
19:22:17 [DEBUG] (1) wchisp::transport::usb: Found USB Device Bus 003 Device 039: ID 1a86:55e0
Error: Other error
```

It helps to power-cycle a connected board.

closed 2024-09-18 by dragonlock2