ch32-rs/wchisp #53: Not detect any device on MACOS

garudaonekh, 2024-07-25:
Hi,
I successfully install libusb but wchisp info can't detect any device.
I use the WCH592F. from WeActStudio.

I use MacOS 13.5

Thanks;

andelf, 2024-07-25:
Could you provide the output of `lsusb` or `system_profiler SPUSBDataType`?

garudaonekh, 2024-07-25:
> system_profiler SPUSBDataType

Here it's

```
2024-07-25 13:32:23.560 system_profiler[95658:10680429] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2024-07-25 13:32:23.561 system_profiler[95658:10680429] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2024-07-25 13:32:23.561 system_profiler[95658:10680429] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2024-07-25 13:32:23.561 system_profiler[95658:10680429] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2024-07-25 13:32:23.562 system_profiler[95658:10680429] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2024-07-25 13:32:23.563 system_profiler[95658:10680429] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2024-07-25 13:32:23.563 system_profiler[95658:10680429] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
USB:

    USB 3.1 Bus:

      Host Controller Driver: AppleUSBXHCITR
      PCI Device ID: 0x15ec 
      PCI Revision ID: 0x0006 
      PCI Vendor ID: 0x8086 
      Bus Number: 0x00 

    USB 3.1 Bus:

      Host Controller Driver: AppleUSBXHCITR
      PCI Device ID: 0x15ec 
      PCI Revision ID: 0x0006 
      PCI Vendor ID: 0x8086 
      Bus Number: 0x01 

    USB 3.1 Bus:

      Host Controller Driver: AppleIntelCNLUSBXHCI
      PCI Device ID: 0xa36d 
      PCI Revision ID: 0x0010 
      PCI Vendor ID: 0x8086 

    Apple T2 Bus:

      Host Controller Driver: AppleUSBVHCIBCE

        Touch Bar Backlight:

          Product ID: 0x8102
          Vendor ID: 0x05ac (Apple Inc.)
          Version: 2.01
          Serial Number: 0000000000000000
          Manufacturer: Apple Inc.
          Location ID: 0x80700000

        Touch Bar Display:

          Product ID: 0x8302
          Vendor ID: 0x05ac (Apple Inc.)
          Version: 2.01
          Serial Number: 0000000000000000
          Manufacturer: Apple Inc.
          Location ID: 0x80600000

        Apple Internal Keyboard / Trackpad:

          Product ID: 0x027c
          Vendor ID: 0x05ac (Apple Inc.)
          Version: 9.27
          Serial Number: FM7833503TWHYYMA6+RCZ
          Manufacturer: Apple Inc.
          Location ID: 0x80500000

        Headset:

          Product ID: 0x8103
          Vendor ID: 0x05ac (Apple Inc.)
          Version: 2.03
          Serial Number: 000000000000
          Manufacturer: Apple
          Location ID: 0x80400000

        Ambient Light Sensor:

          Product ID: 0x8262
          Vendor ID: 0x05ac (Apple Inc.)
          Version: 2.01
          Serial Number: 000000000000
          Manufacturer: Apple Inc.
          Location ID: 0x80300000

        FaceTime HD Camera (Built-in):

          Product ID: 0x8514
          Vendor ID: 0x05ac (Apple Inc.)
          Version: 2.01
          Serial Number: CC28206787EJ3Y311
          Manufacturer: Apple Inc.
          Location ID: 0x80200000

        Apple T2 Controller:

          Product ID: 0x8233
          Vendor ID: 0x05ac (Apple Inc.)
          Version: 2.01
          Serial Number: 0000000000000000
          Manufacturer: Apple Inc.
          Location ID: 0x80100000

```

andelf, 2024-07-25:
It seems that no USB-ISP device was found. Could you check the connection and BOOT pin status?