ch32-rs/wchisp #54: It is possible to use CH375DLL.dll to prevent change driver on Windows.

DeqingSun, 2024-08-20:
Hi, 
Today I came across that the wchisp was [added as the tool for upload](https://github.com/openwch/arduino_core_ch32/commit/1b28a7de76ccc22f139731eb907967074d47429c) in arduino_core_ch32 by Adafruit.

I read the readme and browsed the code (Sorry I never used rust before), it seems the transport is using libUSB only. It is actually possible to check target device on Windows with libUSB first, then fallback to CH375DLL to do communication with WCH driver.

Some of my working code can be viewed on https://github.com/DeqingSun/vnproch551/blob/master/main.cpp

DeqingSun, 2025-01-14:
Ok, as my first try on Rust, I can confirm rust can use CH375DLL64.dll to access bootloader.

```
use libloading::{Library, Symbol};

fn main() {
    println!("Hello, world!");

    unsafe {
        let lib = Library::new("CH375DLL64.dll").expect("Failed to load CH375DLL64.dll");

        let ch375_get_drv_version: Symbol<unsafe extern "C" fn() -> u32> = lib.get(b"CH375GetDrvVersion").expect("Failed to load function");
        let ch375_get_usb_id: Symbol<unsafe extern "C" fn(u32) -> u32> = lib.get(b"CH375GetUsbID").expect("Failed to load function");
        let ch375_open_device: Symbol<unsafe extern "C" fn(u32) -> u32> = lib.get(b"CH375OpenDevice").expect("Failed to load function");
        let ch375_write_data: Symbol<unsafe extern "C" fn(u32, *mut u8, *mut u32) -> bool> = lib.get(b"CH375WriteData").expect("Failed to load function");
        let ch375_read_data: Symbol<unsafe extern "C" fn(u32, *mut u8, *mut u32) -> bool> = lib.get(b"CH375ReadData").expect("Failed to load function");

        let version = ch375_get_drv_version();
        println!("Driver version: {}", version);
        let usb_id = ch375_get_usb_id(0);
        println!("USB ID: 0x{:08x}", usb_id);

        if usb_id == 0x55e04348 || usb_id == 0x55e01a86 {
            println!("CH375 detected");
            let result = ch375_open_device(0);
            if result > 0 {
                println!("CH375 open OK");

                // send detect command
                let mut detect_cmd = [0xA1, 0x12, 0x00, 0x00, 0x11, 0x4D, 0x43, 0x55, 0x20, 0x49, 0x53, 0x50, 0x20, 0x26, 0x20, 0x57, 0x43, 0x48, 0x2e, 0x43, 0x4E];
                let mut write_len = detect_cmd.len() as u32;

                let write_result = ch375_write_data(0, detect_cmd.as_mut_ptr(), &mut write_len);
                println!("Write result: {}", write_result);

                // read response

                let mut detect_respond = [0u8; 64];
                let mut read_len = (detect_cmd[1] + 3) as u32;
                let read_result = ch375_read_data(0, detect_respond.as_mut_ptr(), &mut read_len);
                println!("Read result: {}", read_result);
                println!("Read data: {:02x?}", detect_respond);

            } else {
                println!("CH375 open failed");
            }
        } else {
            println!("CH375 not detected");
        }
    }
}
```

and I got output from CH552.

```
Hello, world!
Driver version: 0
USB ID: 0x55e04348
CH375 detected
CH375 open OK
Write result: true
Read result: true
Read data: [a1, a7, 02, 00, 52, 11, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]
```

Let's see if I can merge the code into WCHISP and get it working without changing driver.

andelf, 2025-01-14:
Yes, it's possible. 

It's implemented in https://github.com/ch32-rs/wlink as a dual-driver backend.

https://github.com/ch32-rs/wlink/blob/217f0e5136892ffcc6a80204a601bc1b62159505/src/usb_device.rs#L161-L168

DeqingSun, 2025-01-14:
Thanks that is great. 

Does [wlink](https://github.com/ch32-rs/wlink) covers everything that [wchisp](https://github.com/ch32-rs/wchisp) can do? Or there are some feature only in [wchisp](https://github.com/ch32-rs/wchisp)?

froscon2024, 2025-03-20:
@DeqingSun One of the examples what wchisp can and wlink cant is flashing firmware to 8bit MCU's: 
https://github.com/ch32-rs/wlink/issues/82

closed 2025-12-09 by DeqingSun