ch32-rs/wchisp #68: WeAct Studio CH32X035F8U6 board

nanikjava, 2024-10-11:
Playing around with the `blink` sample from https://github.com/cnlohr/ch32v003fun/ and managed to compile the code. Tried using `wchisp` to flash but getting `Error: Verify failed, mismatch`.  Flashing procedure uses the following command:

```
wchisp  -v flash ./devboards/ch32v003fun/examples_x035/blink/blink.elf
```

here is the full log 

```
21:40:01 [INFO] Opening USB device #0
21:40:01 [DEBUG] (1) wchisp::transport::usb: Found USB Device Bus 001 Device 103: ID 1a86:55e0
21:40:01 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
21:40:01 [DEBUG] (1) wchisp::transport: <= a1000200 5e23
21:40:01 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
21:40:01 [DEBUG] (1) wchisp::transport: <= a1000200 5e23
21:40:01 [DEBUG] (1) wchisp::flashing: found chip: CH32X035F8U6[0x5e23]
21:40:01 [DEBUG] (1) wchisp::transport: => a70200   1f00
21:40:01 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00ff001fe000ff00ffffffffff00020600cdaba1716cbcdad9
21:40:01 [DEBUG] (1) wchisp::flashing: read_config: ff001fe000ff00ffffffffff00020600cdaba1716cbcdad9
21:40:01 [INFO] Chip: CH32X035F8U6[0x5e23] (Code Flash: 64KiB)
21:40:01 [INFO] Chip UID: CD-AB-A1-71-6C-BC-DA-D9
21:40:01 [INFO] BTVER(bootloader ver): 02.60
21:40:01 [DEBUG] (1) wchisp::transport: => a70200   1f00
21:40:01 [DEBUG] (1) wchisp::transport: <= a7001a00 1f00ff001fe000ff00ffffffffff00020600cdaba1716cbcdad9
21:40:01 [INFO] Current config registers: ff001fe000ff00ffffffffff00020600cdaba1716cbcdad9
RDPR_USER: 0xE01F00FF
  [7:0]   RDPR 0xFF (0b11111111)
    `- Protected
  [16:16] IWDG_SW 0x1 (0b1)
    `- IWDG enabled by the software, and disabled by hardware
  [17:17] STOP_RST 0x1 (0b1)
    `- Disable
  [18:18] STANDBY_RST 0x1 (0b1)
    `- Disable, entering standby-mode without RST
  [20:19] RST_MOD 0x3 (0b11)
    `- Disable RST alternative function, use PA21/PC3/PB7 as GPIO
DATA: 0xFF00FF00
  [7:0]   DATA0 0x0 (0b0)
  [23:16] DATA1 0x0 (0b0)
WRP: 0xFFFFFFFF
  `- Unprotected
21:40:01 [INFO] Read /home/nanik/Downloads/devboards/ch32v003fun/examples_x035/blink/blink.elf as ELF format
21:40:01 [INFO] Found loadable segment, physical address: 0x00000000, virtual address: 0x00000000, flags: 0x5
21:40:01 [DEBUG] (1) wchisp::format: Matching section: ".init" offset: 0x1000 size: 0xe0
21:40:01 [DEBUG] (1) wchisp::format: Matching section: ".text" offset: 0x10e0 size: 0x170
21:40:01 [INFO] Section names: [".init", ".text"]
21:40:01 [DEBUG] (1) wchisp::format: found 1 sections
21:40:01 [INFO] Firmware size: 1024
21:40:01 [INFO] Erasing...
21:40:01 [WARN] erase_code: set min number of erased sectors to 8
21:40:01 [DEBUG] (1) wchisp::transport: => a40400   08000000
21:40:01 [DEBUG] (1) wchisp::transport: <= a4000200 0000
21:40:01 [INFO] Erased 8 code flash sectors
21:40:02 [INFO] Erase done
21:40:02 [INFO] Writing to code flash...
21:40:02 [DEBUG] (1) wchisp::transport: => a31e00   000000000000000000000000000000000000000000000000000000000000
21:40:02 [DEBUG] (1) wchisp::transport: <= a3000200 8600
21:40:02 [DEBUG] (1) wchisp::transport: => a53d00   00000000dc0a65656b656565c303646565016465c365656565016465c365656565656565c301646565016465c365656565656565c301646565656565c3
21:40:02 [DEBUG] (1) wchisp::transport: <= a5000200 0000
█████████░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░ 56/102421:40:02 [DEBUG] (1) wchisp::transport: => a53d00   380000000f01646565656565c301646565016465c301646565016465c301646565016465c301646565016465c301646565016465c301646565016465c3
21:40:02 [DEBUG] (1) wchisp::transport: <= a5000200 0000
███████████████████░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░ 112/102421:40:02 [DEBUG] (1) wchisp::transport: => a53d00   700000001b01646565016465c301646565016465c301646565016465c301646565016465c301646565016465c301646565016465c301646565016465c3
21:40:02 [DEBUG] (1) wchisp::transport: <= a5000200 0000
█████████████████████████████░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░ 168/102421:40:02 [DEBUG] (1) wchisp::transport: => a53d00   a8000000dd01646565016465c301646565016465c301646565016465c301646565016465c301646565016465c301646565016465c301646565656565c3
21:40:02 [DEBUG] (1) wchisp::transport: <= a5000200 0000
███████████████████████████████████████░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░ 224/102421:40:02 [DEBUG] (1) wchisp::transport: => a53d00   e000000040f2646545f6e464b1723465457664e432f667e56d16c567f3c82716f527e5e885726065657660603330e81675305572c665457660408af2c6
21:40:02 [DEBUG] (1) wchisp::transport: <= a5000200 0000
█████████████████████████████████████████████████░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░ 280/102421:40:02 [DEBUG] (1) wchisp::transport: => a53d00   18010000056545f6e0c08b64850633d06575a474c6862bd09b766065e6f2606545f6e0602e726365457663e32f06eda0657124f1027460f46086ffa03d
21:40:02 [DEBUG] (1) wchisp::transport: <= a5000200 0000
███████████████████████████████████████████████████████████░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░ 336/102421:40:02 [DEBUG] (1) wchisp::transport: => a53d00   500100009bd29265856022fd00f662057f16f572f71665455564c552d467257922d263e5c3b0ea79a2d24267836c22fda6d272678352626d6546c762c3
[... 68 lines cut ...]
21:38:45 [INFO] Opening USB device #0
21:38:45 [INFO] 	Device #0: CH32X035F8U6[0x5e23]
21:38:45 [INFO] hint: use `wchisp info` to check chip info
```

Response from `wchisp  config`

```
RDPR_USER: 0xE01F00FF
  [7:0]   RDPR 0xFF (0b11111111)
    `- Protected
  [16:16] IWDG_SW 0x1 (0b1)
    `- IWDG enabled by the software, and disabled by hardware
  [17:17] STOP_RST 0x1 (0b1)
    `- Disable
  [18:18] STANDBY_RST 0x1 (0b1)
    `- Disable, entering standby-mode without RST
  [20:19] RST_MOD 0x3 (0b11)
    `- Disable RST alternative function, use PA21/PC3/PB7 as GPIO
DATA: 0xFF00FF00
  [7:0]   DATA0 0x0 (0b0)
  [23:16] DATA1 0x0 (0b0)
WRP: 0xFFFFFFFF
  `- Unprotected
```

rgoulter, 2024-12-21:
I observed the same output when flashing the WeAct CH32X035 board.

I think the problem is the error message is unclear. e.g. when flashing with the WCHISPStudio, it indicates the error "Read-Protect is enabled,Please disable code-protect first!".

Disabling code-protect requires a WCH-LinkE (or WCH-LinkW). [wlink](https://github.com/ch32-rs/wlink) should be able to do code-unprotect.

nanikjava, 2024-12-22:
> Disabling code-protect requires a WCH-LinkE (or WCH-LinkW). [wlink](https://github.com/ch32-rs/wlink) should be able to do code-unprotect.

@rgoulter Do you know what is the full command to run using `wlink`  ?

rgoulter, 2024-12-22:
With a WCH-LinkE, with the SWCLK and SWDIO pins connected to the CH32X035 (& the CH32X035 powered; e.g. plugged in by USB),

using `wlink` should be as simple as:

```
wlink unprotect
```

Then you can run `wchisp flash` as usual.

However, `wlink flash <path to firmware>` also will also work. Since that doesn't require unplugging/plugging the USB cable (while holding the boot button), it's more convenient than using `wchisp`.