ch32-rs/wchisp #86: Error on the new CH570D-R0-1v0 board

notthetup, 2025-04-10:
Hello. I got my hands on the new CH570D-R0-1v0 from WCH.

[image]

Running `wchisp` on it I get.

```
 ❯ wchisp -V
wchisp 0.3.0
 ❯ wchisp -v info
05:31:13 [INFO] Opening USB device #0
05:31:13 [DEBUG] (1) wchisp::transport::usb: Found USB Device Bus 000 Device 001: ID 1a86:55e0
05:31:13 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
05:31:13 [DEBUG] (1) wchisp::transport: <= a1000200 7013
05:31:13 [DEBUG] (1) wchisp::transport: => a11200   00004d4355204953502026205743482e434e
05:31:13 [DEBUG] (1) wchisp::transport: <= a1000200 7013
Error: Cannot find chip with id 0x70 device_type 0x13
```

I know this is a new board and a new IC. I will be happy to help test any builds/branches.

biemster, 2025-04-22:
People are slowly starting to receive these boards, is anyone working on this?

biemster, 2025-04-22:
> I've managed to flash [@biemster](https://github.com/biemster)'s blink example from [this PR](https://github.com/cnlohr/ch32fun/pull/561) with [some changes to the YAML file](https://github.com/sw/wchisp/commit/1b89d1e6b34f155c589ff3649f10ff4b0d8ddc74).
> 

you mean the blinky actually works? that's awesome!

I'm currently using [chprog.py](https://github.com/wagiminator/MCU-Flash-Tools/issues/10) to flash, but having control of the config bytes using your yaml would be great actually.

dquadros, 2025-04-22:
I've got one of this boards (without the extra chips). Information on them from WCH seems chaotic.

As shown in the photo above, the board is marked as  CH570D-R0-1v0 and the chip as CH570D. Had no luck with the WCH-LinkUtility V2.40 (the latest I could find) and the official WCH-LinkE dongle.

There is a github named https://github.com/openwch/ch570, but its contents refer to a CH572D-R0-1v2 board with a CH572D chip.

biemster, 2025-04-23:
WCH very often pairs two chips in the ch5xx series, with one being a cheaper and stripped down version of the other. In this case 572 is the "full" version and 570 the cheaper stripped version. Confusingly the 570 is marketed the most currently because it tries to take the crown as cheapest RF mcu. And even more confusing there is already a 571/573 pair from a couple years ago.

The linkE should be able to work with the 570 on the latest firmware. There is some amazing reverse engineering of the linkE firmware underway and it has already shown that the 570 starts showing up in there from version 2.16 (my linkE arrived just a couple weeks ago with 2.12)

closed 2025-10-24 by linear[bot]