ch32-rs/wchisp #93: Verify failed with fresh CH592F chips

berkutta, 2025-05-22:
Hello

I have custom boards with new CH592F chips, bought from the WCH AliExpress store.  

If I flash a chip once with WCHISPTool, then it flashes successfully with WCHISP.  
But if it wasn't ever flashed with WCHISP, then I get a verify failed.  
Any idea why?  

Fresh Chip:  
```
✗ wchisp flash makefile-blink.hex
21:52:03 [INFO] Opening USB device #0
21:52:03 [INFO] Chip: CH592[0x9222] (Code Flash: 448KiB, Data EEPROM: 32KiB)
21:52:03 [INFO] Chip UID: 5B-29-8E-10-53-5C-3C-96
21:52:03 [INFO] BTVER(bootloader ver): 02.30
21:52:03 [INFO] Current config registers: ffffffffffffffffd50fff4f000203005b298e10535c3c96
RESERVED: 0xFFFFFFFF
WPROTECT: 0xFFFFFFFF
  [0:0]   NO_KEY_SERIAL_DOWNLOAD 0x1 (0b1)
    `- Enable
  [1:1]   DOWNLOAD_CFG 0x1 (0b1)
    `- PB22(Default set)
USER_CFG: 0x4FFF0FD5
  [2:0]   RESERVED 0x5 (0b101)
    `- Default
  [3:3]   CFG_RESET_EN 0x0 (0b0)
    `- Disable
  [4:4]   CFG_DEBUG_EN 0x1 (0b1)
    `- Enable
  [5:5]   RESERVED 0x0 (0b0)
    `- Default
  [6:6]   CFG_BOOT_EN 0x1 (0b1)
    `- Enable
  [7:7]   CFG_ROM_READ 0x1 (0b1)
    `- Read enable
  [27:8]  RESERVED 0xFFF0F (0b11111111111100001111)
    `- Error
  [31:28] VALID_SIG 0x4 (0b100)
    `- Valid
21:52:03 [INFO] Read makefile-blink.hex as IntelHex format
21:52:03 [INFO] Firmware size: 153600
21:52:03 [INFO] Erasing...
21:52:03 [INFO] Erased 151 code flash sectors
21:52:04 [INFO] Erase done
21:52:04 [INFO] Writing to code flash...
█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████ 153600/15360021:52:04 [INFO] Code flash 153600 bytes written
21:52:05 [INFO] Verifying...
Error: Verify failed, mismatch
```

Already once with WCHISPTool programmed Chip:  
```
✗ wchisp flash makefile-blink.hex
21:52:10 [INFO] Opening USB device #0
21:52:10 [INFO] Chip: CH592[0x9222] (Code Flash: 448KiB, Data EEPROM: 32KiB)
21:52:10 [INFO] Chip UID: 5E-29-8E-10-53-5C-3F-96
21:52:10 [INFO] BTVER(bootloader ver): 02.30
21:52:10 [INFO] Current config registers: ffffffffffffffff4d0fff4f000203005e298e10535c3f96
RESERVED: 0xFFFFFFFF
WPROTECT: 0xFFFFFFFF
  [0:0]   NO_KEY_SERIAL_DOWNLOAD 0x1 (0b1)
    `- Enable
  [1:1]   DOWNLOAD_CFG 0x1 (0b1)
    `- PB22(Default set)
USER_CFG: 0x4FFF0F4D
  [2:0]   RESERVED 0x5 (0b101)
    `- Default
  [3:3]   CFG_RESET_EN 0x1 (0b1)
    `- Enable
  [4:4]   CFG_DEBUG_EN 0x0 (0b0)
    `- Disable
  [5:5]   RESERVED 0x0 (0b0)
    `- Default
  [6:6]   CFG_BOOT_EN 0x1 (0b1)
    `- Enable
  [7:7]   CFG_ROM_READ 0x0 (0b0)
    `- Disable the programmer to read out, and keep the program secret
  [27:8]  RESERVED 0xFFF0F (0b11111111111100001111)
    `- Error
  [31:28] VALID_SIG 0x4 (0b100)
    `- Valid
21:52:10 [INFO] Read makefile-blink.hex as IntelHex format
21:52:10 [INFO] Firmware size: 153600
21:52:10 [INFO] Erasing...
21:52:10 [INFO] Erased 151 code flash sectors
21:52:11 [INFO] Erase done
21:52:11 [INFO] Writing to code flash...
█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████ 153600/15360021:52:13 [INFO] Code flash 153600 bytes written
21:52:13 [INFO] Verifying...
█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████ 153600/15360021:52:14 [INFO] Verify OK
21:52:14 [INFO] Now reset device and skip any communication errors
21:52:14 [INFO] Device reset
```

DeqingSun, 2025-06-19:
Might be the same issue 
https://github.com/ch32-rs/wchisp/issues/34#issuecomment-2830562430
CH573 can not flash with USB with CFG_DEBUG_EN. CH592F might have the same issue.

hykilpikonna, 2026-08-16:
> > CH592F might have the same issue.
> 
> Indeed. Like [#34](https://github.com/ch32-rs/wchisp/issues/34) and [#88](https://github.com/ch32-rs/wchisp/issues/88), the `CFG_DEBUG_EN` bit needs to be unset. I have verified this with my CH592 device.
> 
> --- a/devices/0x22-CH59x.yaml
> +++ b/devices/0x22-CH59x.yaml
> @@ -31,7 +31,7 @@ config_registers:
>    - offset: 0x08
>      name: USER_CFG
>      description: User config register
> -    reset: 0x4FFF0FD5
> +    reset: 0x4FFF0F45
>      type: u32
>      fields:
>        - bit_range: [2, 0]

This worked, thanks!

For people coming here in the future:

1. `git clone https://github.com/ch32-rs/wchisp`
2. Edit `devices/0x22-CH59x.yaml` like the diff from ayu above
3. `cargo install --path .` (make sure you uninstall your previous wchisp so paths wouldn't fight)
4. `wchisp config disable-debug`
5. Run `wchisp config` again to verify that config is indeed `0x4FFF0F45`
6. You're good to flash!

[image]

[image]